opencardev/crankshaft #422: missing dependecies for building openauto in prebuilts (on Rpi4, buster)

egisz, 2020-10-08:
Tried to build prebuilts myself, and faced issue:
Checked out [csng-alpha5.4](https://github.com/matt2005/prebuilts/tree/csng-alpha5.4) prebuilts.
Executed `pi@CRANKSHAFT-NG:~/src/prebuilts/buildsystem/openauto_build $ sudo ./1_prepare_build_system.sh`

But, got errors on step `5_build_openauto.sh` -  missing deps (see bellow): 
I succeed only after manually installing missing packages **without parameter**  `--no-install-recommends` ??? :
`sudo apt-get install librtaudio-dev libtag1-dev libblkid-dev`

```
pi@CRANKSHAFT-NG:~/src/prebuilts/buildsystem/openauto_build $ sudo cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE -DAASDK_INCLUDE_DIRS="$HOME/aasdk/include" -DAASDK_LIBRARIES="$HOME/aasdk/lib/libaasdk.so" -DAASDK_PROTO_INCLUDE_DIRS="$HOME/aasdk_build" -DAASDK_PROTO_LIBRARIES="$HOME/aasdk/lib/libaasdk_proto.so" ../openauto
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   system
--   log
--   unit_test_framework
--   date_time
--   log_setup
--   filesystem
--   thread
--   regex
--   chrono
--   atomic
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
CMake Error at cmake_modules/Findrtaudio.cmake:65 (message):
  Could not find rtaudio
Call Stack (most recent call first):
  CMakeLists.txt:35 (find_package)
-- Configuring incomplete, errors occurred!
```

```
pi@CRANKSHAFT-NG:~/src/prebuilts/buildsystem/openauto_build $ sudo cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE -DAASDK_INCLUDE_DIRS="$HOME/aasdk/include" -DAASDK_LIBRARIES="$HOME/aasdk/lib/libaasdk.so" -DAASDK_PROTO_INCLUDE_DIRS="$HOME/aasdk_build" -DAASDK_PROTO_LIBRARIES="$HOME/aasdk/lib/libaasdk_proto.so" ../openauto
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   system
--   log
--   unit_test_framework
--   date_time
--   log_setup
--   filesystem
--   thread
--   regex
--   chrono
--   atomic
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- Found rtaudio:
--  - Includes: /usr/include/rtaudio
--  - Libraries: /usr/lib/arm-linux-gnueabihf/librtaudio.so
CMake Error at cmake_modules/Findtaglib.cmake:64 (message):
  Could not find taglib
Call Stack (most recent call first):
  CMakeLists.txt:36 (find_package)
```

```
pi@CRANKSHAFT-NG:~/src/prebuilts/buildsystem/openauto_build $ sudo cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE -DAASDK_INCLUDE_DIRS="$HOME/aasdk/include" -DAASDK_LIBRARIES="$HOME/aasdk/lib/libaasdk.so" -DAASDK_PROTO_INCLUDE_DIRS="$HOME/aasdk_build" -DAASDK_PROTO_LIBRARIES="$HOME/aasdk/lib/libaasdk_proto.so" ../openauto
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   system
--   log
--   unit_test_framework
--   date_time
--   log_setup
--   filesystem
--   thread
--   regex
--   chrono
--   atomic
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- Found rtaudio:
--  - Includes: /usr/include/rtaudio
--  - Libraries: /usr/lib/arm-linux-gnueabihf/librtaudio.so
-- Found taglib:
--  - Includes: /usr/include/taglib
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libtag.so;-ltag
CMake Error at cmake_modules/Findblkid.cmake:64 (message):
  Could not find blkid
Call Stack (most recent call first):
  CMakeLists.txt:37 (find_package)
```

egisz, 2020-10-08:
Even after resolving this, build fails:
```
[ 59%] Building CXX object CMakeFiles/autoapp.dir/src/autoapp/Service/ServiceFactory.cpp.o
In file included from /opt/vc/include/interface/vcos/vcos.h:144,
                 from /opt/vc/src/hello_pi/libs/ilclient/ilclient.h:47,
                 from /home/pi/src/prebuilts/buildsystem/openauto/include/f1x/openauto/autoapp/Projection/OMXVideoOutput.hpp:24,
                 from /home/pi/src/prebuilts/buildsystem/openauto/src/autoapp/Service/ServiceFactory.cpp:34:
/opt/vc/include/interface/vcos/vcos_logging.h:234:39: warning: ISO C++ does not permit named variadic macros [-Wvariadic-macros]
 #  define _VCOS_LOG_X(cat, _level, fmt...)   do { if (vcos_is_log_enabled(cat,_level)) vcos_log_impl(cat,_level,fmt); } while (0)
                                       ^~~
In file included from /usr/include/arm-linux-gnueabihf/c++/8/bits/c++allocator.h:33,
                 from /usr/include/c++/8/bits/allocator.h:46,
                 from /usr/include/c++/8/string:41,
                 from /usr/local/qt5/include/QtCore/qbytearray.h:52,
                 from /usr/local/qt5/include/QtCore/qstring.h:50,
                 from /usr/local/qt5/include/QtCore/qcoreapplication.h:44,
                 from /usr/local/qt5/include/QtWidgets/qapplication.h:44,
                 from /usr/local/qt5/include/QtWidgets/QApplication:1,
                 from /home/pi/src/prebuilts/buildsystem/openauto/src/autoapp/Service/ServiceFactory.cpp:19:
/usr/include/c++/8/ext/new_allocator.h: In instantiation of ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = f1x::openauto::autoapp::service::VideoService; _Args = {boo         st::asio::io_context&, std::shared_ptr<f1x::aasdk::messenger::IMessenger>&, std::shared_ptr<f1x::openauto::autoapp::projection::OMXVideoOutput>}; _Tp = f1x::openauto::autoapp::service::VideoService         ]’:
/usr/include/c++/8/bits/alloc_traits.h:475:4:   required from ‘static void std::allocator_traits<std::allocator<_Tp> >::construct(std::allocator_traits<std::allocator<_Tp> >::allocator_type&, _Up*,          _Args&& ...) [with _Up = f1x::openauto::autoapp::service::VideoService; _Args = {boost::asio::io_context&, std::shared_ptr<f1x::aasdk::messenger::IMessenger>&, std::shared_ptr<f1x::openauto::autoa         pp::projection::OMXVideoOutput>}; _Tp = f1x::openauto::autoapp::service::VideoService; std::allocator_traits<std::allocator<_Tp> >::allocator_type = std::allocator<f1x::openauto::autoapp::service::         VideoService>]’
/usr/include/c++/8/bits/shared_ptr_base.h:545:39:   required from ‘std::_Sp_counted_ptr_inplace<_Tp, _Alloc, _Lp>::_Sp_counted_ptr_inplace(_Alloc, _Args&& ...) [with _Args = {boost::asio::io_contex         t&, std::shared_ptr<f1x::aasdk::messenger::IMessenger>&, std::shared_ptr<f1x::openauto::autoapp::projection::OMXVideoOutput>}; _Tp = f1x::openauto::autoapp::service::VideoService; _Alloc = std::all         ocator<f1x::openauto::autoapp::service::VideoService>; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)1]’
/usr/include/c++/8/bits/shared_ptr_base.h:677:16:   required from ‘std::__shared_count<_Lp>::__shared_count(_Tp*&, std::_Sp_alloc_shared_tag<_Alloc>, _Args&& ...) [with _Tp = f1x::openauto::autoapp         ::service::VideoService; _Alloc = std::allocator<f1x::openauto::autoapp::service::VideoService>; _Args = {boost::asio::io_context&, std::shared_ptr<f1x::aasdk::messenger::IMessenger>&, std::shared_         ptr<f1x::openauto::autoapp::projection::OMXVideoOutput>}; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)1]’
/usr/include/c++/8/bits/shared_ptr_base.h:1342:71:   required from ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<f1x::openaut         o::autoapp::service::VideoService>; _Args = {boost::asio::io_context&, std::shared_ptr<f1x::aasdk::messenger::IMessenger>&, std::shared_ptr<f1x::openauto::autoapp::projection::OMXVideoOutput>}; _Tp          = f1x::openauto::autoapp::service::VideoService; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)1]’
/usr/include/c++/8/bits/shared_ptr.h:359:59:   required from ‘std::shared_ptr<_Tp>::shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<f1x::openauto::autoapp::ser         vice::VideoService>; _Args = {boost::asio::io_context&, std::shared_ptr<f1x::aasdk::messenger::IMessenger>&, std::shared_ptr<f1x::openauto::autoapp::projection::OMXVideoOutput>}; _Tp = f1x::openaut         o::autoapp::service::VideoService]’
/usr/include/c++/8/bits/shared_ptr.h:705:14:   required from ‘std::shared_ptr<_Tp> std::allocate_shared(const _Alloc&, _Args&& ...) [with _Tp = f1x::openauto::autoapp::service::VideoService; _Alloc          = std::allocator<f1x::openauto::autoapp::service::VideoService>; _Args = {boost::asio::io_context&, std::shared_ptr<f1x::aasdk::messenger::IMessenger>&, std::shared_ptr<f1x::openauto::autoapp::pro         jection::OMXVideoOutput>}]’
/usr/include/c++/8/bits/shared_ptr.h:721:39:   required from ‘std::shared_ptr<_Tp> std::make_shared(_Args&& ...) [with _Tp = f1x::openauto::autoapp::service::VideoService; _Args = {boost::asio::io_         context&, std::shared_ptr<f1x::aasdk::messenger::IMessenger>&, std::shared_ptr<f1x::openauto::autoapp::projection::OMXVideoOutput>}]’
/home/pi/src/prebuilts/buildsystem/openauto/src/autoapp/Service/ServiceFactory.cpp:81:88:   required from here
[... cut ...]
/home/pi/src/prebuilts/buildsystem/aasdk/include/f1x/aasdk/Channel/AV/IVideoServiceChannelEventHandler.hpp:50:18: note:         ‘virtual void f1x::aasdk::channel::av::IVideoServiceChannelEventHandl         er::onAVChannelStopIndication(const f1x::aasdk::proto::messages::AVChannelStopIndication&)’
     virtual void onAVChannelStopIndication(const proto::messages::AVChannelStopIndication& indication) = 0;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/autoapp.dir/build.make:406: CMakeFiles/autoapp.dir/src/autoapp/Service/ServiceFactory.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/autoapp.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```

egisz, 2020-10-09:
I successfully build aasdk and openauto from [presslab-us](https://github.com/presslab-us/aasdk) sources..
Keypress patch is working fine, crankshaft image also.
Maybe it's worth to make PR from his fork?

matt2005, 2020-10-10:
Hmm, that's odd as i based my prebuilts on the presslab-us ones.
I'm in middle of rebuilding the prebuilts.

matt2005, 2020-10-11:
Can you try my latest prebuilts?
I don't have a pi4 at the moment, so can only check on my pi3.
I've rebuilt the prebuilts in my repo from the latest code in the openauto and aadsk repos.
I've removed the need for the QT compile step as it was clear why it was needed, due to this you need to make sure qt and all the relevant modules are installed.

### In Dev mode, ssh to the pi containing the crankshaft image

- Install qt5
```bash
sudo apt-get -y install libboost-all-dev libusb-1.0.0-dev libssl-dev cmake libprotobuf-dev protobuf-c-compiler protobuf-compiler libqt5multimedia5 libqt5multimedia5-plugins libqt5multimediawidgets5 qtmultimedia5-dev libqt5bluetooth5 libqt5bluetooth5-bin qtconnectivity5-dev pulseaudio librtaudio-dev
```

- Get the new prebuilt binaries
```bash
git clone -b csng-alpha5.4 https://github.com/matt2005/prebuilts.git
sudo cp prebuilts/openauto/autoapp /usr/local/bin/autoapp -f
sudo cp prebuilts/openauto/btservice /usr/local/bin/btservice -f 
sudo cp prebuilts/openauto/libaasdk.so /usr/local/lib/libaasdk.so -f
sudo cp prebuilts/openauto/libaadsk_proto.so /usr/local/lib/libaasdk_proto.so -f
sudo cp prebuilts/gpio2kbd/gpio2kbd /opt/crankshaft/gpio2kbd -f
sudo cp prebuilts/cam_overlay/cam_overlay.bin /opt/crankshaft/cam_overlay/cam_overlay.bin -f
```

egisz, 2020-10-11:
Hi,

I tried, but... it crashes. on rpi4...
When I start it from cmdline, get error:
```
pi@CRANKSHAFT-NG:~ $ /usr/local/bin/autoapp
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.`
```

matt2005, 2020-10-11:
please join so we can discuss
[Slack Channel](https://join.slack.com/t/newworkspace-48q7787/shared_invite/zt-i3f7r9xx-PSfpobCfNErBTGAHWArRhA)

matt2005, 2020-10-12:
I've updated the previous post with the lastest branch

matt2005, 2020-10-15:
Please try this release [alpha 5.5](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha5.5)

matt2005, 2020-10-16:
@egisz Can this now be closed?

closed 2020-10-20 by matt2005