opencardev/crankshaft #423: Wavshare 4.3 Inch Touchscreen Does Not Work

matthewconwa, 2020-10-09:
- What phone do you have?
Xiaomi Mix Mix 3
- What OS version?
9PKQ1.190321.001
- What crankshaft version?
Crankshaft Openauto 0.2.3 2018-04-18
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Start pi -> Connect Phone -> Map Load 

My config: Raspberry Pi 3B + Waveshare 15932 LCD 4,3"

I have a problem with no touch in Raspberry Pi. In Crankshaft settings touch works well but when I plugged the phone and then google maps lunched I lost touch completely. I tried this fix in [click](https://github.com/opencardev/crankshaft/issues/38#issuecomment-373227859) but without any result. Please look at my config and crankshaft_env files. 

Please help.
[crankshaft_env.txt](https://github.com/opencardev/crankshaft/files/5357289/crankshaft_env.txt)
[config.txt](https://github.com/opencardev/crankshaft/files/5357290/config.txt)

matt2005, 2020-10-16:
Please try the latest release

matt2005, 2021-01-07:
Did the screen work on latest release?

matt2005, 2021-11-24:
Closing as no further comments

closed 2021-11-24 by matt2005