opencardev/crankshaft #429: Virtualbox?

PartisanEntity, 2020-10-29:
Hi there,

Do you guys think it would be possible to get Crankshaft running on Virtualbox (on a Windows PC Host)? The idea is to set up an Android Auto test station as easily as possible (obviously as an alternative to installing Crankshaft on dedicated hardware such as a Raspberry Pi).

matt2005, 2020-10-29:
It should be possible in qemu but I've never tried it

parkerlreed, 2020-10-30:
If you just need to test Android Auto functionality for your phone, there's a desktop head unit emulator from Google.

https://developer.android.com/training/cars/testing#install

noharm, 2020-12-04:
I'm currently attempting to run crankshaft on qemu (as I currently dont have a rpi to play with). Based on documentation I found here.
https://azeria-labs.com/emulate-raspberry-pi-with-qemu/ and https://github.com/dhruvvyas90/qemu-rpi-kernel.

I've already figured out I need to modify hardcoded references from /dev/mmcblk0p2 to /dev/sda2. After that is is sort of booting but is getting stuk after resizing the partion.

```
apt install git qemu-system-arm
git clone https://github.com/dhruvvyas90/qemu-rpi-kernel.git
cd qemu-rpi-kernel
wget https://github.com/opencardev/crankshaft/releases/download/csng-alpha7/2020-10-26-crankshaft-ng-f8d7ba3.zip
unzip 2020-10-26-crankshaft-ng-f8d7ba3.zip
losetup -f 2020-10-26-crankshaft-ng-f8d7ba3.img 
kpartx -av /dev/loop0 
mount /dev/mapper/loop0p2 /mnt/
sed -i 's/mmcblk0p2/sda2/g' /mnt/etc/fstab
sed -i 's/mmcblk0p2/sda2/g' /mnt/usr/local/bin/crankshaft
sed -i 's/mmcblk0p2/sda2/g' /mnt/etc/initramfs-tools/scripts/local-top/local.sh
umount /mnt
qemu-system-arm -kernel kernel-qemu-4.19.50-buster -dtb versatile-pb-buster.dtb -cpu arm1176 -m 256 -M versatilepb -serial stdio -append "root=/dev/sda2 rootfstype=ext4 rw" --drive format=raw,file=2020-10-26-crankshaft-ng-f8d7ba3.img -no-reboot
````

noharm, 2020-12-04:
Scratch my previous post (most of it). I've learned the proper kernel and dtb can be easily extracted from the first partition of the image and qemu can emulate the mmcblk0 device.

```
apt install git qemu-system-arm kpartx
wget https://github.com/opencardev/crankshaft/releases/download/csng-alpha7/2020-10-26-crankshaft-ng-f8d7ba3.zip
unzip 2020-10-26-crankshaft-ng-f8d7ba3.zip
losetup -f 2020-10-26-crankshaft-ng-f8d7ba3.img 
kpartx -av /dev/loop0 
mount /dev/mapper/loop0p1 /mnt
cp /mnt/kernel7.img .
cp /mnt/bcm2709-rpi-2-b.dtb .
umount /mnt
qemu-system-arm -kernel kernel7.img -dtb bcm2709-rpi-2-b.dtb -M raspi2 -serial stdio -append "root=/dev/mmcblk0p2 rootwait console=ttyAMA0" --drive format=raw,file=2020-10-26-crankshaft-ng-f8d7ba3.img
```
No solution yet for openauto crashing or missing wifi hardware though.

closed 2021-11-27 by github-actions[bot]