opencardev/crankshaft #431: brightness - lack of reaction WaveShare 15932

matthewconwa, 2020-11-23:
What phone do you have?
Xiaomi Mix Mix 3
What OS version?
9PKQ1.190321.001
What crankshaft version?
Crankshaft OpenCarOS csng-alpha7 2020-10-26 / Android Auto 5.6.603434-release

Hi,
I don't have any idea to run brightness up/down in WaveShare 15932. I connected GPIO pins to port 5,6 and next 22, 27 and I don't have any reaction to press the buttons. I debugged script crankshaft.bash and I explored that I don't have any of those files (/tmp/custombrightness and /sys/class/backlight/rpi_backlight/brightness). I unlocked ssh connection and I created the first file in tmp. Next I ran:
crankshaft brightness set 50
crankshaft brightness save
but my screen doesn't react to any combination in crankshaft.bash i.e (crankshaft display brightness up, crankshaft display brightness down).

Maybe I need some driver. Do you have any idea?
Thanks.

matt2005, 2020-11-27:
I don't have a screen with brightness control but I'll have a look to see if I can figure the issue out.

matt2005, 2020-11-30:
@matthewconwa Is this the brighness modification that you have performed? [Waveshare PWM Brightness control](https://www.waveshare.com/w/upload/5/56/PWM_control_backlight_manual.pdf)

Or is this the screen you have [Waveshare 4.3 HDMI LCD (B)](https://www.waveshare.com/wiki/4.3inch_HDMI_LCD_(B))

matthewconwa, 2020-12-23:
@matt2005 yes, exactly I have a display which you linked. I saw Waveshare PWM Brightness control (really thanks!) but I don't on which port on display I should connect my pad. I don't see my display in this guideline.
This is my model

[image]

matt2005, 2021-01-07:
You could try and figure it out with a logic analyser.

matthewconwa, 2021-01-14:
@matt2005 could you tell me more about this solution " connect one side of each button to a gpio and then ground the connection". Do you have any how-to guides or maybe some examples?

matt2005, 2021-01-14:
I've not got any examples as it was just an idea based on how push buttons work. You will need to check voltage as per post at bottom as you don't want to break the pi. This is something that might work basically you will use the gpio to complete the circuit on the screens button so the screen will think you've pressed the button. [Simulate button press post on stackexchange](https://raspberrypi.stackexchange.com/questions/59060/simulating-button-press)

Hope this helps.

closed 2021-03-12 by matt2005