opencardev/crankshaft #432: Update

KamiSchami, 2020-12-09:
Hi, how do i update a running system?  do I have to reinstall everything?

matt2005, 2020-12-10:
What version are you running at the moment?

KamiSchami, 2020-12-11:
[image]
@

matt2005, 2020-12-20:
It's a major update but if you backup the sd card and then connect pi to internet it should detect an update and you will be able to update. Otherwise you could setup another sd card with the latest release  and then copy over the crankshaft folder and edit config.txt for your setup if required.

matt2005, 2021-01-07:
Did you manage to update?

Lotusvball, 2021-02-10:
I have the same version and my pi is connected to the internet.  It will not update using the Update option in Cranskshaft.

matt2005, 2021-02-10:
@Lotusvball @KamiSchami I've just informed the updater about a new version. I'm also checking that it is picking up the update from your specific version.

matt2005, 2021-02-10:
Try this
1. Enable dev mode
2. SSH into PI
3. Run
```bash
crankshaft update check
```
4. It should say update available
5. Run
```bash
crankshaft update csmt
```
6. turn off dev mode
```bash
crankshaft mode default
```
7. Reboot

matt2005, 2021-02-10:
Please let me know if you experience the pi hanging.

KamiSchami, 2021-02-11:
Ok great, i will test it. Thank you.

Lotusvball, 2021-02-13:
I ended up doing a new image and using that.  It works great.  Much more responsive.

closed 2021-03-05 by matt2005