opencardev/crankshaft #433: Waveshare 7" Touchscreen 1024x600 not working

luizricardo-co, 2020-12-14:
Hi guys, I installed the last release Crankshaft OpenCarOS csng-alpha7 2020-10-26 and not working on LCD 7inch. When I connect on my HDMI monitor works great, but when I connect on my HDMI LDC on show blue screen.
I putted sniping code on my config file and nothing.

hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1

Anybody can help me? 
Thanks!

matt2005, 2020-12-20:
Can you share the config.txt from the sd card

matt2005, 2021-01-13:
I've attached my config.txt which is for a 7" 1024x600 screen, please try it.
[config.txt](https://github.com/opencardev/crankshaft/files/5811177/config.txt)

matt2005, 2021-02-09:
Please try release [alpha7.2](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.2)

closed 2021-03-12 by matt2005