opencardev/crankshaft #435: Crash after touchscreen driver installation

Rizov74, 2020-12-17:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What crankshaft version?
Crankshaft OpenCarOS csng-alpha7 2020-10-26

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
1) Fresh install of Crankshaft on SD card
2) Set Debug mode in crankshaft_env.sh [DEV_MODE=1]
3) Boot on raspberry, log into ssh and install the LCD5-show script from https://github.com/goodtft/LCD-show/blob/master/LCD5-show
4) Reboot with unabled debug mode

And i obtain :
Unfortunatly, OpenAuto crashed
Please report this incident to Crankshaft

I got the crash everytimes, with X11 mode or not

I'm trying to use crankshaft with a Kuman 5 inches tft display : https://www.amazon.fr/gp/product/B01F3JI58G
I also add the lines in the config file. The screen work well, but the touchscreen doens't.

matt2005, 2020-12-20:
Can you share the config.txt from the sd card?

Rizov74, 2020-12-21:
Here the modified config.txt file after touchscreen drivers installation :

[config.txt](https://github.com/opencardev/crankshaft/files/5724547/config.txt)

Basline-LineTech, 2020-12-29:
You don't need a driver for this screen. I have it installed. you just need to put this configs on the config.txt file after a fresh install. 

max_usb_current=1
hdmi_force_hotplug=1
config_hdmi_boost=7
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_drive=1
display_rotate=0
hdmi_cvt 800 480 60 6 0 0 0

matt2005, 2021-01-07:
Yes @Basline-LineTech is correct you shouldn't need to install drivers. It will require a change to the config.txt file as suggested above.

Please test and let us know the outcome.

Rizov74, 2021-01-11:
I've changed the config file after a fresh install, but I still can't use the touchscreen.
I've first tried to only add the changes, the pi booted well, but couldn't use the touchscreen, either in X11 mode or EGL.
After I tried to replace the whole config file with the changes, the pi crashed in EGL mode, and booted in X11 mode, but still without the touchscreen.

Btw, the touchscreen works well on rasberry pi os after the manual install.

I'm probably doing something wrong, sorry for that

Basline-LineTech, 2021-01-11:
Have you tried another version of crankshaft? We need to troubleshoot this.

Rizov74, 2021-01-11:
I tried with the 2020-10-15-csng-alpha5.5-a73be17 version, but without any success.

Wich version should I try ?

And to be clear, after the sd flash, I change :
- the whole config.txt by your changes
- set START_X11=1 in the crankshaft_en.sh file (I read the touchscreen can't be handled in the EGL mode)

Basline-LineTech, 2021-01-12:
You don't need to erase the content of config.txt, just open it and add the text. BTW I use my screen on EGL mode cause mine crashes on X11, maybe you could give it another try. Also, try older versions of crankshaft, like 2019's, cause some things are a bit buggy with those new versions

matt2005, 2021-01-13:
I had issues with incorrect line endings which caused issues so please check that.
@Basline-LineTech Please expand on the buggy comment as I'm attempting to resolve them before adding new features.

Basline-LineTech, 2021-01-13:
Well, I have experienced the same problem with the "new versions" of crankshaft, fortunately I was able to fix it easily with those lines in the config.txt file. Also I can't use Android auto via WiFi hotspot anymore with the new versions, but I'll open another topic for this when I generate the debug.zip file. Also thanks to help us all to troubleshoot our issues

Rizov74, 2021-01-18:
I tested this version without any success :
- 2020-10-26
- 2020-10-15
- 2019-08-08
- 2018-12-21

I didn't pay attention to the end line as matt2005 said, so i retested the versions, not all, only the 2020-10-26 and 2019-08-08

matt2005, 2021-01-18:
What happens without editing the config.txt file.

Rizov74, 2021-01-18:
it's the same, I can't use the touchscreen

Basline-LineTech, 2021-01-18:
I took a closer look and noticed that your screen is slightly different than mine. I have already used some similar to yours and I remember that you do need to install the drivers, and I don't remember how I installed it. Probably you're going to need to connect using SSH, give the required permission to the driver installer and hopefully it will work. I really don't remember how I installed mine 2 years ago, but I remember it was hard to do. You need to install because the touch screen info is sent by the gpio pins and not by usb

Rizov74, 2021-01-18:
It is possible that the gpio pins for the touchscreen are already used by crankshaft ? 
Because when I installed the driver through ssh, it always crashed when I rebooted. 
I wanted to try to run the LCD5-show script line by line to see what cause the crash, but I'm still a newbe in linux env

Basline-LineTech, 2021-01-18:
By default crankshaft doesn't use any gpio pin, but you can configure it to use. About running the script line by line, you can see the entire script using the command `nano LCD5-show.sh`. I'm assuming that this is the name of your file. But after all you probably are gonna need root run all commands successfully, and I don't remember how to do it on crankshaft. Maybe @matt2005 can help us with that?

matt2005, 2021-02-09:
I've just built a new release which fixes some other issues including Debug file generation.
Are you able to try this release and attach a debug file?
I can look at adding support but will need assistance in testing as i don't have the hardware. @Rizov74 @Basline-LineTech 

Please try release [alpha7.2](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.2)

Basline-LineTech, 2021-02-10:
I'll give it a try. But I think showing us how to login as root will be enough since we already have the drivers file.

Rizov74, 2021-02-10:
[debug.zip](https://github.com/opencardev/crankshaft/files/5958205/debug.zip)
Hi,
here the debug files with the alpha7.2 version.
I simply added the lines in the config file, enabled debug mode. 
The touchscreen still doesn't respond

matt2005, 2021-02-10:
Debug mode will enable SSH access, but will still be Readonly, you can use dev_mode [Enable Development Mode](https://github.com/opencardev/crankshaft/wiki/Dev-Mode-and-Debug-Mode#dev-mode)
Once you have it working please provide the steps and we can get it added.

matt2005, 2021-02-11:
Once setup and working please generate a debug file so I can compare and add support

Rizov74, 2021-02-11:
Ok I booted in DEV_MODE and DEBUG_MODE
I generated debug file, and here the result from the scp :/boot/debug.zip command :
[debug.zip](https://github.com/opencardev/crankshaft/files/5965906/debug.zip)

Btw, how can I generate debug files with ssh ? I didn't found the command so I had to click on the gray button.

I'm not sure I understood what exactly you want, if it's not this, can you specify your request ?

matt2005, 2021-02-11:
Is the debug from before you install the driver install?

Rizov74, 2021-02-11:
No I tried to install the drivers in devmode only (if devmode + debugmode, the system remains in RO mode) befor making the debug file.
The debug file I sent yesterday is juste after a fresh install
Here the error :

> pi@CRANKSHAFT-NG:~ $ sudo LCD-show/LCD5-show
> sudo: ./system_backup.sh: command not found
> cp: cannot stat './boot/config-nomal.txt': No such file or directory
> LCD-show/LCD5-show: line 10: ./boot/config.txt.bak: No such file or directory
> grep: ./boot/config.txt.bak: No such file or directory
> sed: can't read ./boot/config.txt.bak: No such file or directory
> sed: can't read ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 15: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 16: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 17: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 18: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 19: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 20: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 21: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 22: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 23: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 24: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 25: ./boot/config.txt.bak: No such file or directory
> LCD-show/LCD5-show: line 26: ./boot/config.txt.bak: No such file or directory
> cp: cannot stat './boot/config.txt.bak': No such file or directory
> cp: cannot stat './usr/cmdline.txt': No such file or directory
> cp: cannot stat './usr/inittab': No such file or directory
> cp: cannot stat './usr/99-fbturbo.conf-HDMI': No such file or directory
> cp: cannot stat './usr/99-calibration.conf-5-0': No such file or directory
> 2021
> need to update touch configuration
> dpkg: error: cannot access archive './xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb': No such file or directory

matt2005, 2021-02-11:
Run this
```bash
crankshaft mode default
crankshaft mode dev
```
Then install the driver and then run the command below.
```bash
crankshaft mode debug
```
To create debug file from ssh run
```bash
crankshaft debuglog
```

matt2005, 2021-02-11:
Also make sure you are running the install from inside the repo folder e.g. cd LCD-SHOW

Rizov74, 2021-02-11:
Ok I'm making a fresh install, I'll come back with debug files.
So step by step :
1) Start once in default mode
2) Enable dev mode by modifying cranckshaft_env on the computer (ssh is not enabled)
3) Start in dev mode
4) install driver, and make sure i'm in the folder LCD-show 
5) Reboot
6) Switch in debug mode with the command 
7) Create log

Btw, if I want to download from git, I have to set my DNS Ip in resolv.conf (It was redirected to localhost)

Rizov74, 2021-02-11:
Ok here the debug files :
[debug.zip](https://github.com/opencardev/crankshaft/files/5966231/debug.zip)

The output when installing drivers :

> pi@CRANKSHAFT-NG:~/LCD-show $ sudo ./LCD5-show
> (Reading database ... 101707 files and directories currently installed.)
> Removing xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
> Processing triggers for man-db (2.8.5-2) ...
> 2021
> need to update touch configuration
> Selecting previously unselected package xserver-xorg-input-evdev.
> (Reading database ... 101697 files and directories currently installed.)
> Preparing to unpack .../xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb ...
> Unpacking xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
> Setting up xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
> Processing triggers for man-db (2.8.5-2) ...
> 
> reboot now

matt2005, 2021-02-12:
try this config.txt
[config.txt](https://github.com/opencardev/crankshaft/files/5969123/config.txt)
The driver install appears to have rememoved settings from the file, so i've added the main one for the screen.

Rizov74, 2021-02-12:
Great the touchscreen works fine now !!

matt2005, 2021-02-13:
No problem, can you upload a debug file so I can see if I can auto detect the screen.

closed 2021-02-13 by matt2005

Rizov74, 2021-02-13:
Here the debug files : 
[debug.zip](https://github.com/opencardev/crankshaft/files/5975875/debug.zip)

I've juste an issue, the touchscreen is offseted and I can't have access to the bottom of the screen.
I can change the calibration with this line right ?
`dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900`

matt2005, 2021-02-13:
those values should be fine. try editing /etc/X11/xorg.conf.d/99-calibration.conf as this will have been installed by the install script from this file https://github.com/goodtft/LCD-show/blob/master/usr/99-calibration.conf-5-0
I don't have the hardware so can't confirm values however the ones in the calibration.conf appear to be larger than the the ones in the dtoverlay line.

scott9000, 2021-03-05:
Not sure if it helps but i think i have the exact same screen, I installed the drivers then copied the config.txt that @matt2005 put here and used the calibration values I took from calibrating the screen in raspbian.
My values were as follows
Option  "Calibration"   "148 3936 229 3976"

MazixM, 2021-07-12:
Hi,
i have same issue as here. I was trying to do everything same as here, but with same result - OpenAuto crash. 
I was trying use Raspberry Pi 4 (I was also thinking about buy RPI 3B.. maybe i have to ?)
- What crankshaft version?
- Crankshaft OpenCarOS NG Alpha7.2 2021-02-09
- Screen, what i am using:
- LCD 5" Waveshare TFT (https://www.waveshare.com/wiki/5inch_HDMI_LCD)
I was using this drivers : https://github.com/waveshare/LCD-show.git
Everything works fine before i install touchscreen driver.
LCD works fine on rasbian with this drivers, so touchscreen is look like ok.
Here is my debuglog. I was trying to find some solution for coupe of days, but without succes. :( 
[debug.zip](https://github.com/opencardev/crankshaft/files/6800575/debug.zip)

Maybe i have to open new issue? I'm new here so please just let me know if i'm doing something wrong.
Have a nice day! :)

Mayur9766, 2024-03-31:
@matt2005 Please could you help me, I'm facing same issue with ri3 and 3.5 ich lcd screen.