opencardev/crankshaft #436: With Alpha7 TSL 2561 stops working

fishred, 2020-12-21:
TS 2561 sensor stops working.

- What OS version?

- What crankshaft version?
 Crankshaft OpenCarOS csng-alpha7 2020-10-26

The issue will disappear if I switch back to the SD-card with image from 2019.

matt2005, 2020-12-22:
Can you compare the config.txt from both versions?

fishred, 2020-12-23:
> Can you compare the config.txt from both versions? I don't have a tsl2561 so cannot test.

Thank you for reminding.
Following is the diff. I had tried the dtparam and dtoverlay for i2c_arm and could not make it work.

>  --- config19.txt	2016-11-03 16:31:46.000000000 +0100
> +++ config22.txt	2020-12-23 12:01:16.131574731 +0100
> @@ -43,15 +43,23 @@
>  #arm_freq=800
>  
>  \# Uncomment some or all of these to enable the optional hardware interfaces
> -dtparam=i2c_arm=on
>  #dtparam=i2s=on
>  #dtparam=spi=on
>  
> -# Uncomment this to enable the lirc-rpi module
> -#dtoverlay=lirc-rpi
> +# Uncomment this to enable infrared communication.
> +#dtoverlay=gpio-ir,gpio_pin=17
> +#dtoverlay=gpio-ir-tx,gpio_pin=18
>  
>  \# Additional overlays and parameters are documented /boot/overlays/README
>  
> +[pi4]
> +# Enable DRM VC4 V3D driver on top of the dispmanx display stack
> +dtoverlay=vc4-fkms-v3d
> +max_framebuffers=2
> +
> +[all]
> +dtoverlay=vc4-fkms-v3d
> +
>  \# Custom power settings
>  max_usb_current=1
>  
> @@ -68,6 +76,15 @@
>  \# Overscan fix
>  overscan_scale=1
>  
> +# Enable watchdog
> +dtparam=watchdog=on
> +
> +# Boot time improvements
> +boot_delay=0
> +initial_turbo=30
> +start_cd=1
> +#dtoverlay=sdtweak,overclock_50=100
> +
>  \# Display rotation
>  lcd_rotate=2
>  
> @@ -79,4 +96,6 @@
>  
>  
>  \# I2C Bus
> +dtparam=i2c_arm=on
>  dtoverlay=i2c_arm=on
> +
> 
>

fishred, 2020-12-23:
Actually, from logging output on console, i can see the tsl 2561 has been detected and checked. it only does not work.

matt2005, 2021-01-07:
I'm waiting to receive the hardware so I can test. I'd guess it's an issue with the upgrade to Buster. Have you looked at the wiki page for the TSL 2591? [TSL 2591](https://github.com/opencardev/crankshaft/wiki/TSL2591)

fishred, 2021-01-08:
@matt2005, I tried them, but none of them worked. And I checked my SD card with old version of crankshaft. it is also using 0x39.

matt2005, 2021-01-08:
I'll have a look over the weekend, I don't have the hardware yet but will check that it's being called, as it might not be executing.

matt2005, 2021-01-13:
I think it may be related to a missing package, (smbus)
Try 
```
sudo apt install python-smbus
```

or
```
sudo pip3 install smbus
```

matt2005, 2021-01-14:
@fishred I'm also working on a fix for tsl2591, which will enable both tsl2561 and tsl2591. Please see #437 if you can help test the new script it'll be great.
Thanks

fishred, 2021-01-15:
@matt2005  i copied service_i2ccheck.sh and add following lines to crankshaft_env.sh
`LIGHTSENSOR_TYPE='TSL2561'
TSL_I2C_BUS=1
TSL_ADDR=0x39`

The console output in booting shows device has been detected at that address, but sensor seems not working.
Maybe I missed anything?

matt2005, 2021-01-15:
You need to update [/opt/crankshaft/service_lightsensor.py](https://github.com/opencardev/crankshaft/blob/lightsensor_fixes/stage3/03-crankshaft-base/files/opt/crankshaft/service_lightsensor.py)

I can't test as I've not got the hardware.

fishred, 2021-01-15:
@matt2005  do I need to install python-tsl2591 although I am using tsl2561?

matt2005, 2021-01-15:
Yes, i would as its a module that it loaded by the python script. @fishred

fishred, 2021-01-16:
I have problem to run apt command because I got no network connection now. I am using LAN.

fishred, 2021-01-16:
After figured out my own network issue, I had installed listed component.
The TSL2561 tab in the first round of boot disappeared. After I pick it again, the TSL 2561 sensor seems still not working.

Console output in booting shows sensor has been found.

matt2005, 2021-01-16:
Did you run all the commands including the one for tsl2591 module?

fishred, 2021-01-16:
Yes. tsl2591 Python module had been installed. Very likely that is why the first boot, tsl2561 disappeared from tab.

matt2005, 2021-01-17:
Can you stop the service lightsensor and then run the file manually. Does it run ok? It should return values from the sensor.

fishred, 2021-01-22:
Sorry, it was a busy week. @matt2005, what is the command line to do that?

matt2005, 2021-01-27:
i can't remember it at the top of my head, I'll check when I get my new Pi4

matt2005, 2021-02-06:
@BodoMinea @fishred please try [DEV release alpha 7.1](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.1)

fishred, 2021-02-07:
With this build, it works after my changing default address from 0x29 to 0x39.
Thanks for all the effort. Cool!

fishred, 2021-02-08:
Base on fix from above, i close the issue. Thanks again.

closed 2021-02-08 by fishred