opencardev/crankshaft #437: Using TSL2591 lux sensor as per wiki instructions, getting warning on boot

BodoMinea, 2021-01-10:
Hi, I have followed the wiki instructions (https://github.com/opencardev/crankshaft/wiki/TSL2591) on how to use a TSL2591 instead of a TSL2561 because that was far easier to get a hold of from a local store in my area. 

I haven't properly tested it in my car installation yet but from the get-go I have the following issue. With everything connected and done according to the wiki (also scanned for I2C addresses and I found 0x29 as a match, this being what the TSL2591 uses), I get the following message when Crankshaft-NG boots:

Check for TSL2561 failed. Device at 0x39 is missing!

I guess this check is hard-coded somewhere else. I tried searching with grep for mentions of TSL2561 or 0x39 but to no avail.

I have also ticked the „Show Lux value from tsl2561 on Main Window” in the TSL2561 tab of the Crankshaft settings, but that doesn't do anything for me.

Is there a way to also change the boot-check in order to not have warnings? It is nagging me a bit to see them everytime.

Thanks

BodoMinea, 2021-01-10:
Fixed with PR on the script repo. Needed pip3 smbus module and comparison fix for values between levels 4 and 5. See https://github.com/nomath4u/crankshaft_tsl2591/issues/1 and https://github.com/nomath4u/crankshaft_tsl2591/pull/2

matt2005, 2021-01-13:
@nomath4u please can you submit a pr to add this to the main release.

matt2005, 2021-01-13:
I've make a quick edit to enable support for both tsl2561 and tsl2591. 

I have added a couple of extra configuration options in [/opt/crankshaft/crankshaft_default_env.sh](https://github.com/opencardev/crankshaft/blob/d1cc04e159fde60a7e742f3b573bc3f60d6e0da7/stage3/03-crankshaft-base/files/opt/crankshaft/crankshaft_default_env.sh#L190)
And Updated [/opt/crankshaft/service_lightsensor.py](https://github.com/opencardev/crankshaft/blob/lightsensor_fixes/stage3/03-crankshaft-base/files/opt/crankshaft/service_lightsensor.py)

I can't test as I've not got the hardware.
@BodoMinea Are you able to test?
 If you add the 2 files above and add a similar entry to below in /boot/crankshaft/crankshaft_env.sh

```bash
# LightSensor
LIGHTSENSOR_TYPE = 'TSL2561' # Allowed Values TSL2561 TSL2591
# the address of TSL2561/TSL2591 can be
# 0x29, 0x39 or 0x49
TSL_I2C_BUS = 1
TSL_ADDR = 0x29
```

BodoMinea, 2021-01-14:
Unfortunately, it is not working. Here is what I tried:

After copying the new `service_lightsensor.py` and setting the `LIGHTSENSOR_TYPE`, `TSL_I2C_BUS` and `TSL_ADDR`, upon executing the script, it was complaining about conversion of int values and unrecognized commands.

- The int conversion problems were because there was int conversion in `service_lightsensor.py` in get_var, but the LIGHTSENSOR_TYPE is a string. So I removed the int conversion from the return and applied it manually to `daynight_gpio`, `TSL_I2C_BUS `and `TSL2561_CHECK_INTERVAL`. This eliminated this error.

- Then I edited  `crankshaft_env.sh` and removed spaces around the variables:

```
LIGHTSENSOR_TYPE='TSL2591'
TSL_I2C_BUS=1
TSL_ADDR=0x29
```

Now there is no error but also the script has no output (console logs) and the brightness shown on screen does not change, day/night does not change etc. It is just blank.

BodoMinea, 2021-01-14:
Okay, I think I fixed it. More conversion errors from get_var. See PR #440

Also, do you know how to hide the boot warning about TSL2561?

matt2005, 2021-01-14:
@BodoMinea Thanks for your help, It's really helpful havibg somone witht he hardware to test things, I've only recently taken over to maintenance of Crankshaft so I've not go a lot of the hardware. 
I've merged the PR and updated the default config to reflect your change.
I've also made a fix for the boot warning to enable it to support the tsl2591 and use the configured addresses in the config.

Replace this file /opt/crankshaft/service_i2ccheck.sh with this one [/opt/crankshaft/service_i2ccheck.sh](https://github.com/opencardev/crankshaft/blob/lightsensor_fixes/stage3/03-crankshaft-base/files/opt/crankshaft/service_i2ccheck.sh)

BodoMinea, 2021-01-15:
Yes, that worked like a charm. There's no red warning and the detection works fine, and as such I will close this issue.

I think the next steps would be adding an option to select the kind of the light sensor in the UI and also update the wiki.

Thanks!

closed 2021-01-15 by BodoMinea