opencardev/crankshaft #438: Using RCA video output with touchscreen input

BodoMinea, 2021-01-10:
Hi,

I still haven't got all my components delivered but if it is not a problem I want to ask „in advance”. Has anyone used RCA video output with touchscreen support?

My stock head unit supports RCA input with an aftermarket harness, so I will be using a Raspberry Pi 4 with the latest release (csng-alpha7 2020-10-26 / commit f8d7ba3) and `enable_tvout=1` in config.txt.

I see people around here suggesting to also add the following to config.txt for touchscreen support:
```
max_usb_current=1
hdmi_force_hotplug=1
config_hdmi_boost=7
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_drive=1
display_rotate=0
hdmi_cvt 800 480 60 6 0 0 0
```
However, these parameters seem to be based on the assumption of HDMI output, which in my case it is unfortunately not true.

I am waiting for the following touchscreen to arrive:

https://www.aliexpress.com/item/32976239800.html?spm=a2g0s.9042311.0.0.27424c4d9DkuKt (variant 6.2-A, that's the one that matches the size and resolution of my stock nav display) and the I2C - USB converter from the same listing.

What are your thoughts on this? Is it better (as in... easier / more plug and play) to attach the device to Raspberry's I2C (where I already have the RTC and a TSL2591 lux sensor connected) or to use the USB converter? Is there a way to set it up „HDMI-independently” as opposed to the instructions above?

Thanks!

matt2005, 2021-01-13:
Some people have used an hdmi to RCA converter as they say the resolution is better. However it's not something that I've tried. In theory the touch interface is normally usb so it shouldn't matter if the display is rca.

egisz, 2021-01-13:
I am using native Raspberry RCA output and video quality is usable, because screen isn't too big.
I'ts 10" screen, but due car's stupid settings it only occupies part of it 4:3 format with black borders around.

I believe these settings are to match resolution (screen coordinates to input), but I may be wrong.

matt2005, 2021-02-09:
Please try release [alpha7.2](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.2)

BodoMinea, 2021-02-13:
My update is that I got the touchscreen and USB interface for it and it just worked on my alpha7.1 installation with RCA output. No configuration was required. I just have `enable_tvout=1`.

closed 2021-02-13 by BodoMinea