opencardev/crankshaft #441: Problem rotate waveshare 7,9 inch with the last version cranshaft

Gwx76, 2021-01-21:
Hello,
I can't seem to do a 90 ° rotation on my waveshare screen.

I tried several methods in the config.txt files but without result

I got there very simply with the previous version, why?

Thank you for your reply

crankshaft version: csng-alpha7 2020-10-26

[config.txt](https://github.com/opencardev/crankshaft/files/5851006/config.txt)

matt2005, 2021-01-21:
What previous version where you using? Alpha6?

matt2005, 2021-02-07:
please try [DEV release alpha 7.1](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.1)

Gwx76, 2021-02-09:
hello, i just installed the latest version of crankshaft (DEV release alpha 7.1),  90% of my problems are solved. I am happy with this version because it is by far the best of all the one I tested. 
Thank you very much!

PS (my phone is Xiaomi Mi note 10)  
with raspberry PI 3 B  and screen 7.9" waveshare

matt2005, 2021-02-09:
@Gwx76 what are the other issues? Have you rasied them?

Gwx76, 2021-02-09:
I have a problem with the connection between the smartphone's wifi and crankshaft.
the crankshaft hotpoint is working but the client mode does not. So i can't use wifi for android auto

matt2005, 2021-02-09:
Ok I have a fix for that I'm just working on a new release

Gwx76, 2021-02-09:
ok good job Matt2005!

matt2005, 2021-02-09:
@Gwx76 Please try release [alpha7.2](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.2)

Gwx76, 2021-02-10:
I just tested the 7.2 crankshaft with the patch but I still cannot activate the wifi.
 in the BT / WiFi tab, I activate Client mode but I cannot enter the SSID because everything is erased before I can register.

matt2005, 2021-02-10:
Try following this 
[Wifi Setup](https://github.com/opencardev/crankshaft/wiki/Android-Auto-Wifi)
IF you want to connect to your phones hotspot you need to use WIFI2 as shown in [Connect to Phone](https://github.com/opencardev/crankshaft/wiki/Android-Auto-Wifi#connecting-to-your-phones-wifi-hotspot)

Ensure you also force update for wifi [wifi force update](https://github.com/opencardev/crankshaft/wiki/Android-Auto-Wifi#forcing-an-update)
Then reboot

Gwx76, 2021-02-10:
I use Client mod  and I follow this "Pixel 4XL cant connect via Wifi #389" 

_matt2005 commented 2 days ago
Please try following
Use latest dev release (currently alpha7.1)
Enable mobile hotspot on Phone,
Configure WIFI2 in /boot/crankshaft/crankshaft_env.sh
Set wifi_enable_update=1 in /boot/crankshaft/crankshaft_env.sh
Start crankshaft
Click on wifi connect button, it will error saying unable to connect
on phone start AA and then start headunit server
On Crankshaft click connect
Phone should beep and then AA should open in crankshaft._

 I don't get the "connection refused" error message but AA does not start??

thank for your support

matt2005, 2021-02-10:
please attach debug logs

Gwx76, 2021-02-10:
[debug.zip](https://github.com/opencardev/crankshaft/files/5962309/debug.zip)

matt2005, 2021-02-10:
I think it's failing as you've specified the same SSID for both.  Try setting the first SSID to a different SSID and set WIFI_UPDATE_CONFIG=1
```text
# Wifi client mode (use this entry for home/work wifi - not for phone's hotspot)
# If your SSID or password contains special chars or spaces make sure using quotation marks ="SSID" / ="password"
WIFI_SSID="sample"
WIFI_PSK=xxx cleared xxx
# Optinal 2nd config to access to phone's hotspot
# Note: if second config is configured system will use first connectable wifi during boot
WIFI2_SSID="Mi note 10"
WIFI2_PSK=xxx cleared xxx

# Force recreate wpa_supplicant.conf during boot after wifi credentials have changed by user (default 0 | update 1)
WIFI_UPDATE_CONFIG=1
```

Gwx76, 2021-02-11:
the solution was there
 I just did the test and AA in wifi starts up without problem thank you very much!!!!

I will have one last question but if that doesn't suit the main topic.
can the system time stay up to date during a prolonged shutdown?

matt2005, 2021-02-11:
you'll need a RTC (Real Time Clock) for that.  see [RTC guide](https://www.abelectronics.co.uk/kb/article/30/rtc-pi-on-raspbian-buster-and-stretch)

Gwx76, 2021-02-11:
I will try this RTC thanks Matt2005

matt2005, 2021-02-11:
Issue resolved

closed 2021-02-11 by matt2005

Pavlo458, 2022-02-05:
Hello, I seem to have a similar issue. I've tried alpha 7, 7.1 and 7.2 and cannot rotate my waveshare 7.9 display on any of them. It is by default in portrait view and I want to have it in landscape.
Do you know what I should try?

[config.txt](https://github.com/opencardev/crankshaft/files/8008448/config.txt)