opencardev/crankshaft #442: Screen Not Responding In Night Mode

Ch3ddar77994, 2021-01-25:
When I turn night mode on (using the button in the menu) the screen goes black. I assumed that was normal and that touching the screen would turn it back on, but the when I touch the screen nothing happens. I have tried connecting my phone when this happens and I get the connected to android auto notification (on my phone, a Samsung Note 9) , but the screen is still black and not responding. I am using the official raspberry pi 7 inch touch screen and a pi 3 b+.

matt2005, 2021-02-03:
What version are you using? How are you connecting the phone?

Ch3ddar77994, 2021-02-04:
I am using the newest version and connecting via a usb a to usb c cable.

matt2005, 2021-02-07:
Can you try this version [DEV release alpha 7.1](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.1)
please generate a debug file and attach

matt2005, 2021-09-29:
See #495

closed 2021-09-29 by matt2005