opencardev/crankshaft #443: Crankshaft doesn't recognize AA Headless Server

deejaypro, 2021-02-02:
- What phone do you have?
OnePlus 7 Pro
- What OS version?
Android 10
- What crankshaft version?
Latest (Alpha7)
- What Raspberry Version?
Raspberry Pi 4 2GB
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
I run crankshaft in clientmode. I enabled to see the network information on the main screen.
Everytime I connect crankshaft to my phone via my phones hotspot, it doesn't recognize that the headless server on my phone is running. So the connect button on crankshaft always says "No USB Device" an doesn't change.
Additionally to that, the network gateway information stays blank on the main screen.

- Please provide any further information that you might find helpful if available.
Currently this is my test setup at my desk. I have a working unit in my car which runs on alpha 5 or 6 with a workaround to get the touchscreen working again inside AA.
If i connect my phone via USB everything works fine.
- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

Additionally it seems that i cant produce a debug.zip.
It says "Creating debug.zip" on the main screen but i cant find any debug.zip inside the boot directory via a ssh session..,

matt2005, 2021-02-03:
I'll take a look at my setup as this is how I use it.

deejaypro, 2021-02-05:
Regarding my last point, not being able to gernate a debug.zip.
Just saw the following in journalctl:
```
Feb 05 14:54:26 CRANKSHAFT-NG sudo[7742]:       pi : TTY=unknown ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/chmod 666 /tmp/debuglogs/alsactl.state /tmp/debuglogs/alsa_soundcards_kernel /tmp/debuglogs/amixer.output /tmp/debuglogs/autoapp_helper_command /tmp/debuglogs/c
Feb 05 14:54:26 CRANKSHAFT-NG sudo[7742]:       pi : (command continued) /tmp/debuglogs/set_soundcard /tmp/debuglogs/startsequence.svg /tmp/debuglogs/syslog /tmp/debuglogs/system.pa /tmp/debuglogs/timezone /tmp/debuglogs/usbdevices.log /tmp/debuglogs/user.log /tmp/debugl
Feb 05 14:54:26 CRANKSHAFT-NG sudo[7742]: pam_unix(sudo:session): session opened for user root by (uid=0)
Feb 05 14:54:27 CRANKSHAFT-NG sudo[7742]: pam_unix(sudo:session): session closed for user root
Feb 05 14:54:27 CRANKSHAFT-NG sudo[7744]:       pi : TTY=unknown ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/rm -rf /boot/debug.zip
```
Doesn't the last line mean, that the debug.zip was immediately deleted after creating it?
I tried to find the script generating the logs in this repository but couldn't find it.

matt2005, 2021-02-05:
Thanks - I've added the zip package to the packages for the next release.

matt2005, 2021-02-05:
I'm going to release a version over this weekend once I've checked a few things.

matt2005, 2021-02-05:
what happens when you run crankshaft in hotspot mode and attach your phone to Crankshaft?

deejaypro, 2021-02-06:
Should the SSID be broadcasted or is it a hidden network?

matt2005, 2021-02-06:
its broadcast but you need to turn on hotspot mode inthe settings.
I've created a dev release with the debug fix.
please try [DEV release alpha 7.1](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.1)

deejaypro, 2021-02-06:
Then the hotspot mode didn't work either. It wouldn't broadcast the SSID nor would it let me connect if I treated it as a hidden network.
But for some reason I see some DHCP offers for my phone in the dnsmasq.log. (see [debug__.zip](https://github.com/opencardev/crankshaft/files/5936623/debug__.zip)). Note this is with the old release. 
I will try both modes with the new dev version and I will let you know how it went.

deejaypro, 2021-02-06:
Okay I tested the new version with a few different setting. Good thing: I got it to work in hotspot mode. 
I observed a few different behaviors:
- In client mode it will never recognize the headless server running on my phone: [debug_client.zip](https://github.com/opencardev/crankshaft/files/5936733/debug_client.zip)
- When first starting into Hotspot mode the SSID wont be broadcasted: [debug_hotspot_not_broadcasting.zip](https://github.com/opencardev/crankshaft/files/5936735/debug_hotspot_not_broadcasting.zip)
- When switching from hotspot into client mode the pi will get a ip address in the range 169.254.0.0/16 which isn't that useful: [debug_169address.zip](https://github.com/opencardev/crankshaft/files/5936736/debug_169address.zip)
- When starting in hotspot mode then switching into client mode and then switching back into hotspot mode the SSID gets broadcasted and i can connect to the hotspot without any issues an then I can run AA on the unit: [debug_169_then_succesfull_hotspot.zip](https://github.com/opencardev/crankshaft/files/5936740/debug_169_then_succesfull_hotspot.zip)

I can get the hotspot mode working every time if I switch between the different modes. It doesn't matter if I start in hotspot or client mode. After switching to client mode in the setting it works.

I hope everything is understandable and if you have any question don't hesitate to ask. 
If you need further testing let me know how I can help you.

matt2005, 2021-02-08:
can you check the pi can communicate to the phone head uint service but running.
```
nc -vz 192.168.1.189 5277
```
Replace 192.168.1.189 with your phones IP.

deejaypro, 2021-02-08:
Yeah Sure:
pi@CRANKSHAFT-NG:~ $ nc -vz 192.168.43.83 5227
nc: connect to 192.168.43.83 port 5227 (tcp) failed: Connection refused

A ping to my Phone (43.83) works.
So ist seems ist cant communitcate with the Server...

matt2005, 2021-02-08:
When in client mode and network info shown on main screen. Does it show yours phones ip as the gateway?

matt2005, 2021-02-08:
Are you running the head unit server on your phone?

deejaypro, 2021-02-08:
[image]

Nope GW is emtpy ( and the Ssid is showing the wrong one).
And my bad, sorry. Misstyped the Port:
CRANKSHAFT-NG:~ $ nc -vz 192.168.43.83 5277
Connection to 192.168.43.83 5277 port [tcp/*] succeeded!

matt2005, 2021-02-08:
Try setting the wifi_update=1 in the crankshaft_env config.

deejaypro, 2021-02-08:
Just did that. Still the same:
```
CRANKSHAFT-NG:~ $ nc -vz 192.168.43.222 5277
Connection to 192.168.43.222 5277 port [tcp/*] succeeded!
```
GW is still emtpy and the wrong ssid ist just a visual bug when changing from WLAN 1 to WLAN 2.

matt2005, 2021-02-08:
Can you try a swap the wifi settings around? So the primary is the phone?

deejaypro, 2021-02-08:
Sure no problem.
I completely removed my Home WiFi to minimize the potential "risk" of running into a Bug.
But the results are the same:
```
@CRANKSHAFT-NG:~ $ nc -vz 192.168.43.203 5277
Connection to 192.168.43.203 5277 port [tcp/*] succeeded!
```
[image]

matt2005, 2021-02-08:
I'll step through the code later but I think its failing to detect the gateway ip correctly. I know I can force the gateway, i just can't remember all the steps so I'll check and update later.

deejaypro, 2021-02-08:
Sounds great. If you need further testing just let me know.
I tried to find the code where the gateway is detected earlier to see if i can spot something but i couldn't pinpoint it to a specific part of the code.

deejaypro, 2021-02-08:
I found the line [Line 21](https://github.com/opencardev/crankshaft/blob/5b228516a2448c72fa90f5b79948940d12e4b4ce/stage3/03-crankshaft-base/files/opt/crankshaft/service_wpachange.sh#L21).

Maybe its because of Buster or the Raspberry Pi4 (more likely Buster) the file doesnt exist under "/var/run/dhcpcd/resolv.conf/wlan0.dhcp". Instead it is located at "/var/run/resolvconf/interfaces/wlan0.dhcp".

If i change that line of code the GW is shown inside the GUI. Doing some further tests right now.

matt2005, 2021-02-08:
i'll have a look in a bit but.
If you try the following, add your phone ip into the file.
```
sudo nano /tmp/gateway_wlan0
``

deejaypro, 2021-02-08:
I mean that's what the code does and after changing the path in the if clause my phone could connect to the AA headless server.

matt2005, 2021-02-08:
can you submit a PR to fix the filepath.

PR #446 (2021-02-08): Changed path to check if the headunit got a dhcp lease
Changed the path to check if the user got a dhcp lease.
This should fix problems to start AA on the headunit when running it in Client mode.
Should fix #443
closed 2021-02-08 by matt2005