opencardev/crankshaft #447: OpenAuto crashed with RPi 3A+

Compizfox, 2021-02-13:
- What phone do you have?
N/A

- What crankshaft version?
Alpha7.2 2021-02-09

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Flash the image to a SD card, and boot it.

After boot, I get the following error:

```
Unfortunately, OpenAuto crashed.
Please report this indicident to Crankshaft
Restart in 5 seconds...
```

I tried both EGL and X11. I'm using a HDMI screen.

I'm not quite sure how to create a debug log, since I don't get far enough to push the gray button on bottom of main screen". I tried by just booting in debug mode anyway but I don't get any log.

matt2005, 2021-02-13:
I've not tried a RPi 3A+, but it should work but it will need debugging as it's not a hardware config i've tested.

[Enable Dev_Mode](https://github.com/opencardev/crankshaft/wiki/Dev-Mode-and-Debug-Mode#how-to-enable) 
login to ssh

To create debug file from ssh run
```text
crankshaft debuglog
```

matt2005, 2021-02-17:
@Compizfox Can you try the latest build?

Compizfox, 2021-02-27:
[debug.zip](https://github.com/opencardev/crankshaft/files/6054945/debug.zip)

Here you go.

scott9000, 2021-03-01:
I am also trying to run crankshaft on rpi 3a+ and getting the same issue. Would love to find out if this gets resolved 🙂

matt2005, 2021-03-01:
@scott9000 Can you create a debug file? please ensure that you are not in x11 mode.

scott9000, 2021-03-01:
Yeah will do. Will send it to you when I'm finished work. Thank you!

scott9000, 2021-03-02:
[debug.zip](https://github.com/opencardev/crankshaft/files/6066226/debug.zip)
let me know if this is what you need. I'm a bit of a noob but can follow instrucions

matt2005, 2021-03-02:
@scott9000 what is your setup? What attached devices? Screen? Etc

scott9000, 2021-03-02:
I have a 5" elecrow LCD touch screen and pixel 4, the screen is working
fine but no GUI

Compizfox, 2021-03-02:
> @scott9000 Can you create a debug file? please ensure that you are not in x11 mode.

The debug log I submitted was while running in X11 mode. Do you need one in EGL mode from me as well?

matt2005, 2021-03-02:
@Compizfox The logs appear to show crashing. Can you try editing the gpu_mem in /boot/config.txt. Try editing the gpu_mem from 256 to 128. 
```bash
# GPU Mem
gpu_mem=128
```

Compizfox, 2021-03-04:
Thanks, that did the trick! OpenAuto is now starting correctly.

matt2005, 2021-03-04:
do you have Androidauto installed on Phone? You might need to connect via a powered usb hub, as the pi 3a might not have enough power over usb.
Alternatively try wireless mode see wiki for details

Compizfox, 2021-03-04:
> do you have Androidauto installed on Phone?

Yes, I installed Android Auto and when I connect it, it says "connected to car" on the phone. However, nothing happens on the OpenAuto head unit.

> You might need to connect via a powered usb hub, as the pi 3a might not have enough power over usb.

Hmm, I'm afraid I don't have one on hand. Why does it need to power the phone anyway? I'm not expecting it to charge it or anything, lol

>  Alternatively try wireless mode see wiki for details

Which wiki page exactly? My first idea was to try Bluetooth, but I couldn't find out how to pair it. The [Bluetooth wiki page](https://github.com/opencardev/crankshaft/wiki/Bluetooth) says to "tap the `BT-Pairing` button` but I don't see that...

matt2005, 2021-03-05:
[AA wireless](https://github.com/opencardev/crankshaft/wiki/Android-Auto-Wifi)
 You will need to enable the onboard BT

Enable onboard BT. You must reboot after this.
```bash
crankshaft bluetooth builtin
sudo reboot
```

matt2005, 2021-03-12:
Did the above work for you? @Compizfox

Compizfox, 2021-03-27:
@matt2005 No, unfortunately not.

I was able to enable Bluetooth and pair my phone, but that appears to be for audio only?

I tried wireless Android Auto mode over WiFi, but that doesn't work either. When I connect my phone to Crankshaft device over WiFi it seems to detect it (the top-left button gains a Android Auto icon, but when I press on it nothing happens.

Compizfox, 2021-04-27:
I think this issue can be closed.

Thanks again for the help!

closed 2021-04-27 by Compizfox