opencardev/crankshaft #448: SSD for boot Crankshaft

Gwx76, 2021-02-17:
- What phone do you have Xiaomi Note 10

- What OS version Android

- What crankshaft version 7.2

hello, would it be possible to boot an SSD instead of a mirco SD? allow a faster start? Has someone already tried?

matt2005, 2021-02-17:
Surely that's a waste of a SSD as the file system is read only. 
You can attach ssd as usb media to store media files.
I'm working on a boot time speed up, but currently  concentrating on bug fixing.

Gwx76, 2021-02-18:
Ok no problem, I thought the SDD read small files faster than a mrico SD. but I am not a specialist.

modemlamer, 2021-02-18:
On PI4 its no problem to boot from SSD attached tu USB3

https://www.tomshardware.com/how-to/boot-raspberry-pi-4-usb

On PI3 it is also possible but harder to achieve

https://www.youtube.com/watch?v=ubnwvxF3Klc

It depends on your SD-CARD if its worth it ... 
the SD-Cardreader is still on USB2.0 as far as i know, so USB3 could be faster
SSD Drives are also more durable... but there are also tough SD-Cards on the market...

in any case => a SSD consumes much more Power!! take care of it