opencardev/crankshaft #450: RPi 2 Model B v1.1 (ARMv7) is not supported

fneon, 2021-03-23:
Problem:
Wiki states that the RPi2 is supported.
This is partially true:
there is at least one version of RPi2 which is not supported: RPi2 v1.1 ARMv7

`
Oct 26 22:30:15 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Oct 26 22:30:15 CRANKSHAFT-NG pi: SERV-LOGGER: Caller: /opt/crankshaft/service_openauto.sh
Oct 26 22:30:15 CRANKSHAFT-NG pi: SERV-LOGGER: Starting OpenAuto in EGL Mode
Oct 26 22:30:15 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Oct 26 22:30:15 CRANKSHAFT-NG systemd[1]: btservice.service: Main process exited, code=killed, status=4/ILL
Oct 26 22:30:15 CRANKSHAFT-NG systemd[1]: btservice.service: Failed with result 'signal'.
Oct 26 22:30:16 CRANKSHAFT-NG service_openauto.sh[971]: /opt/crankshaft/service_openauto.sh: line 182:  1357 Illegal instruction     stdbuf -i0 -o0 -e0 /usr/local/bin/autoapp >> /tmp/openauto.log
Oct 26 22:30:16 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Oct 26 22:30:16 CRANKSHAFT-NG pi: SERV-LOGGER: Caller: /opt/crankshaft/service_openauto.sh
Oct 26 22:30:16 CRANKSHAFT-NG pi: SERV-LOGGER: Openauto crashed
Oct 26 22:30:16 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
`

Investigation showed that Qt binaries are built for the ARMv8 platform:
`
pi@CRANKSHAFT-NG:~ $ readelf -A /usr/local/qt5/lib/libQt5Widgets.so.5
Attribute Section: aeabi
File Attributes
  Tag_CPU_name: "8-A"
  Tag_CPU_arch: v8
  Tag_CPU_arch_profile: Application
  Tag_ARM_ISA_use: Yes
  Tag_THUMB_ISA_use: Thumb-2
  Tag_FP_arch: FP for ARMv8
  Tag_Advanced_SIMD_arch: NEON for ARMv8
  ...
`

matt2005, 2021-03-26:
I'll look into kicking off a qt rebuild on a pi2 when I dig it out.

Jasoroony, 2021-04-01:
I don't want to go back and do any of it again, it was a real pain, but I ended up doing Pi-ZeroW, Pi3 & Pi4 builds, all on my Pi4-4GB with small changes to the config for each, which ended up being much faster and less error-prone than on the device it was targeting.  The Pi-ZeroW didn't finish well, I recall, it ran for days, paged a lot, and I gave up on it - except it all worked OK when I re-did it on my Pi4 with lots of memory that didn't need a pagefile with the Pi0's config.

Corwin616, 2021-11-19:
please fix it

techdev5521, 2021-11-19:
If @Jasoroony or @matt2005 are willing to help me understand the process I am happy to help with this. Possibly even look into QEMU setups for desktop builds.

Jasoroony, 2021-11-20:
I followed this guide (it's for QT5.10.1 which you can't download anymore, but I just changed it to use 5.12.4):

[https://github.com/MarkusIppy/QT5.x-raspbian-stretch](url)

There are qmake.conf files there for the Pi2 & Pi3 - this is the one I created for the Pi0W:

```
# qmake configuration for the Raspberry Pi

include(../common/linux_device_pre.conf)

QMAKE_RPATHLINKDIR_POST += $$[QT_SYSROOT]/opt/vc/lib

QMAKE_LIBDIR_OPENGL_ES2 = $$[QT_SYSROOT]/opt/vc/lib
QMAKE_LIBDIR_EGL        = $$QMAKE_LIBDIR_OPENGL_ES2
QMAKE_LIBDIR_OPENVG     = $$QMAKE_LIBDIR_OPENGL_ES2

QMAKE_INCDIR_EGL        = $$[QT_SYSROOT]/opt/vc/include \
                          $$[QT_SYSROOT]/opt/vc/include/interface/vcos/pthreads \
                          $$[QT_SYSROOT]/opt/vc/include/interface/vmcs_host/linux
QMAKE_INCDIR_OPENGL_ES2 = $${QMAKE_INCDIR_EGL}
QMAKE_INCDIR_OPENVG     = $${QMAKE_INCDIR_EGL}

QMAKE_LIBS_EGL 			= -lbrcmEGL -lbrcmGLESv2
QMAKE_LIBS_OPENGL_ES2 	= -lbrcmEGL -lbrcmGLESv2
QMAKE_LIBS_OPENVG 		= -lbrcmEGL -lbrcmOpenVG -lbrcmGLESv2

QMAKE_CFLAGS            += --target=arm-eabi -marm -mfpu=vfp -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -mfloat-abi=hard
QMAKE_CXXFLAGS          = $$QMAKE_CFLAGS

DISTRO_OPTS += hard-float
DISTRO_OPTS += deb-multi-arch

# Preferred eglfs backend
EGLFS_DEVICE_INTEGRATION = eglfs_brcm

include(../common/linux_arm_device_post.conf)

load(qt_config)
```

These are the various changes to the configure command (in the guide) I had tried to get it working (top one is what I ended up using for the Pi0W):

```
#echo "Configure QT "
PKG_CONFIG_LIBDIR=/usr/lib/arm-linux-gnueabihf/pkgconfig PKG_CONFIG_SYSROOT_DIR=/ \../qt-everywhere-src-5.12.4/configure -v -opengl es2 -eglfs -no-xcb -no-xcb-lib  -no-pch -no-gtk                     -device linux-rpi-g++          \-device-option CROSS_COMPILE=/usr/bin/ -opensource -confirm-license -reduce-exports \-force-pkg-config -nomake examples -no-compile-examples -skip qtwayland -skip qtwebengine -release \-qt-pcre -ssl -evdev -system-freetype -fontconfig -glib -gstreamer -prefix /opt/QT5
PKG_CONFIG_LIBDIR=/usr/lib/arm-linux-gnueabihf/pkgconfig PKG_CONFIG_SYSROOT_DIR=/ \../qt-everywhere-src-5.10.1/configure -v -opengl es2 -eglfs                      -no-pch -no-gtk                     -device linux-rasp-pi2-g++     \-device-option CROSS_COMPILE=/usr/bin/ -opensource -confirm-license -reduce-exports \-force-pkg-config -nomake examples -no-compile-examples -skip qtwayland -skip qtwebengine -release \-qt-pcre -ssl -evdev -system-freetype -fontconfig -glib -gstreamer -prefix /opt/QT5
PKG_CONFIG_LIBDIR=/usr/lib/arm-linux-gnueabihf/pkgconfig PKG_CONFIG_SYSROOT_DIR=/ \../qt-everywhere-src-5.12.4/configure -v -opengl es2 -eglfs                      -no-pch -no-gtk -no-use-gold-linker -device linux-rasp-pi3-vc4-g++ \-device-option CROSS_COMPILE=/usr/bin/ -opensource -confirm-license -reduce-exports \-force-pkg-config -nomake examples -no-compile-examples -skip qtwayland -skip qtwebengine -release \-qt-pcre -ssl -evdev -system-freetype -fontconfig -glib -gstreamer -prefix /opt/QT5
PKG_CONFIG_LIBDIR=/usr/lib/arm-linux-gnueabihf/pkgconfig PKG_CONFIG_SYSROOT_DIR=/ \../qt-everywhere-src-5.12.4/configure -v -opengl es2 -eglfs -no-xcb -no-xcb-xlib -no-pch -no-gtk                     -device linux-rasp-pi3-g++     \-device-option CROSS_COMPILE=/usr/bin/ -opensource -confirm-license -reduce-exports \-force-pkg-config -nomake examples -no-compile-examples -skip qtwayland -skip qtwebengine -release \-qt-pcre -ssl -evdev -system-freetype -fontconfig -glib -gstreamer -prefix /opt/QT5
PKG_CONFIG_LIBDIR=/usr/lib/arm-linux-gnueabihf/pkgconfig PKG_CONFIG_SYSROOT_DIR=/ \../qt-everywhere-src-5.12.4/configure -v -opengl es2 -eglfs -no-xcb -no-xcb-xlib -no-pch -no-gtk                     -device linux-rasp-pi3-vc4-g++ \-device-option CROSS_COMPILE=/usr/bin/ -opensource -confirm-license -reduce-exports \-force-pkg-config -nomake examples -no-compile-examples -skip qtwayland -skip qtwebengine -release \-qt-pcre -ssl -evdev -system-freetype -fontconfig -glib -gstreamer -prefix /opt/QT5
PKG_CONFIG_LIBDIR=/usr/lib/arm-linux-gnueabihf/pkgconfig PKG_CONFIG_SYSROOT_DIR=/ \../qt-everywhere-src-5.12.4/configure -v -opengl es2 -eglfs -no-xcb -no-xcb-xlib -no-pch -no-gtk                     -device linux-rasp-pi4-v3d-g++ \-device-option CROSS_COMPILE=/usr/bin/ -opensource -confirm-license -reduce-exports \-force-pkg-config -nomake examples -no-compile-examples -skip qtwayland -skip qtwebengine -release \-qt-pcre -ssl -evdev -system-freetype -fontconfig -glib -gstreamer -prefix /opt/QT5
```

Don't ask me what excluding xcb does (or why I ended up using -no-xcb-lib instead of -no-xcb-xlib) because it was too long ago now!  I also don't remember why there's a no gold-linker in there, but I did read about it.  The Pi4 one may have been me hacking together a new device for Qt5.12.4 from newer sources, really not sure though.

I'm (not so?) confident you just configure it on the device you want (Pi0W in my case), move the SD card to the Pi4 and "make" (build) it on the faster hardware.  But I had to run the build process a lot of times before it worked, so that makes old memories merge and hard to remember, but I do recall doing it on the Pi4 hardware was just so much better.

And this is the end of any help I can provide, sorry.

matt2005, 2021-11-20:
Yeah the issue is that QT needs recompiling, it's not a simple task, as when it was recompiled it for the pi4 that would have been when the pi2 support was lost.

I'll dig out my pi2 and kick off a compile job, from memory it took about 48hours.

techdev5521, 2021-11-22:
@matt2005 Would you be willing to document that process for me so I can learn? If you give me even a run down of commands and basic explanations I would be happy to flesh out a beginner friendly version of the process for docs.

matt2005, 2021-11-23:
@techdev5521 In a nut shell as i understand it QT is compiled in order to support eglfs so we can run without X. When the disk image is built it pulls the precompiled binaries from the prebuilts repo. The prebuilts repo contains the compile scripts in the buildsystem directory to build it all. I suspect when I upgraded from Stretch to Buster and recompiled for PI4 support it removed the pi2 support.

matt2005, 2021-12-08:
Currently compiling on pi2 should have something working by weekend. Let me know if you are able to test.

Corwin616, 2021-12-08:
@matt2005 i will ready to test

matt2005, 2021-12-09:
Try this from dev mode.
This will remove qt5 and replace it with the version i've just built for a pi2 (arm7). Once this is done you should be able to reboot and it should work.

```bash
rm -rf /usr/local/qt5
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part00
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part01
cat Qt_5151_armv7l_OpenGLES2.tar.xz* > Qt5_OpenGLES2.tar.xz
tar -xf Qt5_OpenGLES2.tar.xz -C /
```

Corwin616, 2022-01-02:
@matt2005 sorry for the delay, there was a lot of work, a business trip.
I downloaded the latest crankshaft img, downloaded and merged your archive, copied it to boot, turned on dev mode, booted from raspberry, removed qt5 folder and unpacked the archive to the root.
then I turned off dev mode and rebooted.
the graphical interface did not load.
how to enable display of detailed logs to see the error?

dasmikko, 2022-01-03:
I also have a RPi2 1.1 that i would love to use for this. Is there any way i can help too?

matt2005, 2022-01-08:
i'm rebuilding it as it silently errored with some of the modules so they are missing. I'm in process of sourcing a rpi2 of the correct revision to assist with the resolution

matt2005, 2022-01-08:
I've got it working, I've rebuild qt5 and updated the config.txt.
Update QT
```
rm -rf /usr/local/qt5
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part00
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part01
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part02
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part03
cat Qt_5151_armv7l_OpenGLES2.tar.xz* > Qt5_OpenGLES2.tar.xz
tar -xf Qt5_OpenGLES2.tar.xz -C /
```

Here is my working config.txt

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi2]

[pi3]
# Bluetooth
dtoverlay=pi3-disable-bt
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

# Custom power settings
max_usb_current=1

# Disable the PWR LED.
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

# Disable Rainbow splash
disable_splash=1

# GPU Mem
gpu_mem=128

# Overscan fix
overscan_scale=1

# Enable watchdog
dtparam=watchdog=on

# Boot time improvements
boot_delay=0
initial_turbo=30
start_cd=1
#dtoverlay=sdtweak,overclock_50=100

# Bluetooth
#dtoverlay=pi3-disable-bt
```

dasmikko, 2022-01-10:
@matt2005 Tested your fix this morning, and sadly it still crashes for me on my RPi 2 v1.1

Tried using the latest crankshaft release.

matt2005, 2022-01-10:
What is the output for your pi when you run the following
```bash
 cat /proc/cpuinfo
```

Here is my output, (I've only removed the serial number)
```text
pi@CRANKSHAFT-NG:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a21041
Serial          : HIDDEN
Model           : Raspberry Pi 2 Model B Rev 1.1
```

dasmikko, 2022-01-10:
> What is the output for your pi when you run the following
> 
> ```shell
>  cat /proc/cpuinfo
> ```
> 
> Here is my output, (I've only removed the serial number)
> 
> ```
> pi@CRANKSHAFT-NG:~ $ cat /proc/cpuinfo
> processor       : 0
> model name      : ARMv7 Processor rev 5 (v7l)
> BogoMIPS        : 57.60
> Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xc07
> CPU revision    : 5
> 
> processor       : 1
> model name      : ARMv7 Processor rev 5 (v7l)
> BogoMIPS        : 57.60
> Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xc07
> CPU revision    : 5
> 
> processor       : 2
> model name      : ARMv7 Processor rev 5 (v7l)
> BogoMIPS        : 57.60
> Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xc07
> CPU revision    : 5
> 
> processor       : 3
> model name      : ARMv7 Processor rev 5 (v7l)
> BogoMIPS        : 57.60
> Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xc07
> CPU revision    : 5
> 
> Hardware        : BCM2835
> Revision        : a21041
> Serial          : HIDDEN
> Model           : Raspberry Pi 2 Model B Rev 1.1
> ```

Here is the output:

```
pi@CRANKSHAFT-NG:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a01041
Serial         
Model           : Raspberry Pi 2 Model B Rev 1.1
pi@CRANKSHAFT-NG:~ $
```

matt2005, 2022-01-10:
I've just written a crahs diag page
can you try the following [crash diag](https://github.com/opencardev/crankshaft/wiki/Crash-diagnosis)

matt2005, 2022-01-10:
Here are the steps I've taken.

1. Download crankshaft image
2. Fash SD card
3. Enable dev mode by updating /boot/crankshaft/crankshaft_env.sh DEV_MODE=1
4. Update /boot/config.txt
5. Boot pi
6. Edit /etc/resolv.conf and update nameserver to reflect correct DNS server ip
```bash
sudo nano /etc/resolv.conf
```  
7. Update QT
```bash
sudo rm -rf /usr/local/qt5
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part00
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part01
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part02
wget https://github.com/opencardev/prebuilts/raw/master/qt5/Qt_5151_armv7l_OpenGLES2.tar.xz.part03
cat Qt_5151_armv7l_OpenGLES2.tar.xz* > Qt5_OpenGLES2.tar.xz
sudo tar -xf Qt5_OpenGLES2.tar.xz -C /
```
8. Run autoapp it crashed
```bash
export QT_DEBUG_PLUGINS=1
GST_DEBUG=3 /usr/local/bin/autoapp -platform eglfs
```
9. Link needed libraries
```bash
sudo ln -s /opt/vc/lib/libbrcmEGL.so /usr/lib/arm-linux-gnueabihf/libEGL.so
sudo ln -s /opt/vc/lib/libbrcmGLESv2.so /usr/lib/arm-linux-gnueabihf/libGLESv2.so
sudo ln -s /opt/vc/lib/libbrcmOpenVG.so /usr/lib/arm-linux-gnueabihf/libOpenVG.so
sudo ln -s /opt/vc/lib/libbrcmWFC.so /usr/lib/arm-linux-gnueabihf/libWFC.so
```
10. Run Autoapp
```bash
export QT_DEBUG_PLUGINS=1
GST_DEBUG=3 /usr/local/bin/autoapp -platform eglfs
```

Corwin616, 2022-01-11:
@matt2005 crankshaft work for me, thank you!

dasmikko, 2022-01-11:
@matt2005 She boots now!

matt2005, 2022-01-11:
Good to hear, I'll look at building it into the next release.
If possible Please sign up as Pateron if you find this useful as I use the money to purchase devices to add support.

dasmikko, 2022-01-11:
Just a fyi, your fix also made it run on a raspberry pi zero 2 W!

And it actually runs pretty well too.

TildenWinston, 2022-02-06:
For what it is worth I followed the above instructions but when trying to start the openauto server I was met with the following error. 
`Got keys from plugin meta data ("eglfs_kms")
QFactoryLoader::QFactoryLoader() checking directory path "/usr/local/bin/egldeviceintegrations" ...
loaded library "/usr/local/qt5/plugins/egldeviceintegrations/libqeglfs-brcm-integration.so"
* failed to add service - already in use?
QLibraryPrivate::unload succeeded on "/usr/local/qt5/plugins/egldeviceintegrations/libqeglfs-brcm-integration.so"
QLibraryPrivate::unload succeeded on "/usr/local/qt5/plugins/platforms/libqeglfs.so"`

Looking into it this seems to be an OpenGL related issue. I was able to get around it by enabling X11 mode in crankshaft_env.sh

matt2005, 2022-02-06:
You've got the fkms driver loading, you need to comment out the following line under [all].
```
dtoverlay=vc4-fkms-v3d
```

F4u5t, 2022-03-17:
I got it working with X11, still errors otherwise...these command aborts on me
GST_DEBUG=3 /usr/local/bin/autoapp -platform eglfs

closed 2022-09-11 by github-actions[bot]