opencardev/crankshaft #452: External GPS receiver support (for better reception/positioning)

BodoMinea, 2021-03-31:
Hi,

Is there a way to feed data into Android Auto from an external GPS transceiver (for example a U-Blox module that transmits GPS NMEA messages by UART - Serial). I am able to get this data onto the Raspberry Pi, but I do not know if it is possible to somehow feed it towards Android Auto.

Stock car units and supposedly Open Auto Pro can do this (https://bluewavestudio.io/community/showthread.php?tid=735). 

It would really help with accuracy, because I am keeping my phone down near the armrest and the location sometimes lags behind. 

I am kindly waiting for the input of someone who tinkered with this matter before.

Thanks!

matt2005, 2021-03-31:
@BodoMinea It was added previously by someone else and I've not looked at the setup.All the pieces should already be available but I'm not sure how its put together, as no documentation exists.
1. gpsd is installed in the image already.
2. support exists in Crankshaft and aasdk
It might be visiable in gui once gpsd sees a device?

BodoMinea, 2021-04-01:
Okay, I will try configuring my GPS receiver with gpsd as that´s pretty straightforward / standard. Do you have any idea of the source file where this is implemented or a commit?

I tried searching around and the only mention is this https://github.com/opencardev/crankshaft/blob/aa60ef188602653c0a1fbba56ffb4144722e24d9/stage3/03-crankshaft-base/02-run-chroot.sh it does some setup but no feeding of data.

parkerlreed, 2021-04-01:
gpsd comms normally are just over local TCPIP. 

Try setting up the GPS for gpsd and verifying data with cgps. Then see if crankshaft gets a "better" position.