opencardev/crankshaft #453: nokia 7 plus

MichelKoch, 2021-04-01:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?

- What OS version?

- What crankshaft version?

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

- Please provide any further information that you might find helpful if available.

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

matt2005, 2021-04-01:
You've not added information.

MichelKoch, 2021-04-02:
Sorry, I was lost here. I do not have a bug but was looking for support. Thx

closed 2021-04-02 by MichelKoch