opencardev/crankshaft #473: GPIO "Android PIN" not doing anything.

CGFlyer85, 2021-06-23:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
 : Redmi Note 8 and 9, Nokia 8

- What OS version?
 : Android 10

- What crankshaft version?
 : Alpha 7.2

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
 : Just set the "Android PIN" GPIO pin to any of the available, plug your phone and, nothing. This bug is always present and on any pin.

- Please provide any further information that you might find helpful if available.
 : First thing, thanks for this wonderful piece of software that I have been using for the past 2 year !
I wanted to instal a new unit with a pop-up screen when android auto is launched, so I saw a GPIO was available for that. Sadly it doest do anything for me, and I dont see any command on the debug file...
Any help will be realy apreciated. Thanks in advance.

(sorry for my bad english as I'm a french native speaker ^^')

- Starting with pre4: Boot into dev mode (can be selected in settings).
  Reproduce the possible bug. Create the debug logs by pushing the grey button on bottom of screen.
  Add the created file (/boot/debug.zip) to the issue.

[debug.zip](https://github.com/opencardev/crankshaft/files/6704268/debug.zip)

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

matt2005, 2021-06-24:
I'm not sure what you are attempting to do. Please add the link to what you are doing and add screenshot if possible.

CGFlyer85, 2021-06-25:
> I'm not sure what you are attempting to do. Please add the link to what you are doing and add screenshot if possible.

Hello, as I understand it from the wiki and a past issue, it work as an output pin wich get anabled when android auto is running.
Correct me if I'm wrong ^^' 
I'm just trying to get a signal when Android Auto is running.

The setting page from the wiki :
[image]

And the previous issue, wich was fixed, but is maybe broken now ?
#302 

Thanks for your help

CGFlyer85, 2021-07-07:
Hello, Any ideas ?

matt2005, 2021-07-08:
I'm looking into it.

matt2005, 2021-09-21:
I've found the issue and am working on the fix.

matt2005, 2021-09-23:
pleaase try alpha 7.4

matt2005, 2021-09-29:
@CGFlyer85 have you been able to retest?

CGFlyer85, 2021-09-30:
Hi @matt2005. Thanks for your Update.
Sorry I dint had time to test, lot of thing going on lately ^^'
I'll try next week and keep you updated. ;)

matt2005, 2021-11-24:
Release 7.4 fixes this. Please reopen and attach debug if not.

commit fe7e2716 (2021-09-22): added wiringpi fix #484 fix #474 fix #473 on pi4
closed 2021-11-24 by matt2005