opencardev/crankshaft #474: Cannot select GPIO27 for "ignition pin"

Compizfox, 2021-07-04:
As per [the wiki page](https://github.com/opencardev/crankshaft/wiki/Boot,-reboot-and-shutdown-the-Pi-with-ignition-key), I made a circuit to pull GPIO27 low when the key is removed from the ignition.

However, in the settings menu the select box for "ignition pin" goes from 1 to 26, thus I'm not able to select GPIO27.

matt2005, 2021-07-04:
i'll take a look at it

Stimpson-J-Cat, 2021-08-05:
I had this set up on a 3B+. I had a power supply set up on the accessory circuit in my truck that stayed powered for 7 min after shutdown. The Pi was connected to that. I had a relay set up that powered with ignition and set GPIO3 low (to ground from pin 9) and that would start the Pi, then when I shut the truck down, the relay would release, ground would drop off GPIO3 and the Pi would shut down, then 7 min later, the power supply would lose power and then no more draw. That 3B+ died and I replaced it with a 4B and this set up no longer works. I even modified the wiring on my relay to match the Wiki and set GPIO3 low when ignition on and GPIO27 low when ignition off and all that happens is the Pi will boot and 10 seconds later, it shuts down (10 seconds is the delay I have set). It seems that any change in the state of GPIO3 makes the Pi boot; if I set it to low, it boots then shuts down in 10, and if I remove the ground from GPIO3 it boots again then shuts down in 10 seconds.  I also had a similar relay set up to GPIO12 to dim the display when the lights were on and that no longer works on the Pi 4B either. Am I better off going back to a 3B+ and not using the 4B?

matt2005, 2021-09-21:
I've found the issue and am working on the fix.

matt2005, 2021-11-24:
Release 7.4 fixes GPIO, you should be able to manually enter it in the config file. Please reopen if it doesn't work.

commit fe7e2716 (2021-09-22): added wiringpi fix #484 fix #474 fix #473 on pi4
closed 2021-11-24 by matt2005