opencardev/crankshaft #485: Crankshaft use wireless like carsifi

LeSchurke, 2021-09-16:
is it possible to use crankshaft wirless like carsifi.
Or is it possible with the hotspot function by crankshaft?

matt2005, 2021-09-18:
Whats carsifi?

LeSchurke, 2021-09-19:
It's an Adapter. 
He is connect to android car by a normal phone cable and your phone by Bluetooth and wireless. 
Now, your phone is automatically connect with Android car and you don't need a phone cable. 

https://carsifi.com/

matt2005, 2021-09-19:
you can skip the initial cable
https://github.com/opencardev/crankshaft/wiki/Android-Auto-Wifi

LeSchurke, 2021-09-19:
I connect to crankshaft via the hotspot and start the Android auto server. The Android auto settings screen popped up at the beginning, but before I could confirm anything, I was suddenly on the phone's homescreen and crankshaft doesn't switch to Android auto mode.
Nothing happens at all. I can't even press the client connect button in crankshaft anymore.

LeSchurke, 2021-09-20:
I don't know why, but now it works.

closed 2021-09-20 by LeSchurke