opencardev/crankshaft #495: Night mode and bigness issues Consolidating 467 and 442 in to one and adding more context

tompeel, 2021-09-29:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have - pixel 4XL

- What OS version - 11

- What crankshaft version - 7.2 & 7.4 on PRi3 and 4

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible. - reproducible every time

- Please provide any further information that you might find helpful if available.

When switching between night mode and day mode, the screen UI updates to the new state, but the brightness, if set to less than 100 is so dark you cant see anything at all on the screen, if you push the place where the day button is, all comes back fine with a glitch.

The glitch seems to be in the control of the brightness power its self, it flashes in and out twice normally and then come back ok (assuming you have the brightness above 100)
This is different behaviour to my built of 12 months ago. 
I assume there may have been some update to the PWM driver perhaps which could be causing issues where the range isn't mapped correctly.

the range less than 100 being useless isn't so much the issue as the flashing in and out is, it looks very hardware interface and I have done this on 2 different pies with 2 different official pi 7inch screens

matt2005, 2021-09-29:
So day/night mode is activated, but the brightness control is not working via the slider or via day/night mode?

matt2005, 2021-09-29:
Please can you provide a debug log from both pi3 and pi4

tompeel, 2021-09-30:
Will try to get a Debug log today.
But yes, the software is all seemingly happy under the hood, just the dimming of the screen goes too far and effectively turns the screen off.... (on a separate note that's a nice feature so if you park up for under an hour you could leave CS booted and turn off the screen to save loading time and battery) 

It very much feels like Whats going on is the number (seems to be less than 110 on the brightness scale) is mapping to low for the screen to function, a fix might be to just limit the lowest number but the flash in and out is also leading me to think there might just need to be an update to a dependency or something for the GPIO.

Logs to follow

matt2005, 2021-09-30:
My suspiscion is that the backlight is turning off, i don't the same screen so I'm going to have to rely on you for testing.

matt2005, 2021-11-25:
appears to be related to #515

closed 2022-03-25 by github-actions[bot]