opencardev/crankshaft #497: Pi-top and Crankshaft

vedranius, 2021-10-05:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Samsung Galaxy S9
- What OS version?
Latest
- What crankshaft version?
Latest
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
I'm using Pi-top case and their display.
When I use Crankshaft and Pi-top together, it boots up fine, but reboots after 10 seconds and goes to loop.
When I'm using Crankshaft without the Pi-top case it works OK.
Any idea what might be causing this?
I don't have time to get the debug log because it starts to reboot...
Tnx!

matt2005, 2021-10-05:
Try [Debug Mode](https://github.com/opencardev/crankshaft/wiki/Dev-Mode-and-Debug-Mode)

vedranius, 2021-10-07:
@matt2005 is there a way to install Crankshaft to Pi-Top OS?
Because in that way I don't need your image, and I can still use all the features from Pi-Top OS.
Is that possible?
Any steps how to do that?

Tnx!

matt2005, 2021-10-09:
Crankshaft is the OS.