opencardev/crankshaft #515: Various issues with lightsensor.service using TSL2561

techdev5521, 2021-11-23:
### What phone do you have?
Google Pixel 5

### What OS version? / What crankshaft version?
Alpha 7.4

### Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
- _Results vary according to unknown factors._
- Connect TSL2561 to Pi

| TSL2561 Pin | | Pi Pin |
| - | - | - |
| VCC | `-->` | Pin 1 (3.3v) |
| GND | `-->` | Pin 9 (GND) |
| SCL | `-->` | Pin 5 (SCL) |
| SDA | `-->` | Pin 3 (SDA) |

- Enable light sensor in settings: Settings `-->` CS HW `-->` Lightsensor = tsl2561
- (Optional) Enable Lux output on screen: Settings `-->` TSL2561 `-->` Check 'Show Lux value....'
- Restart

### Please provide any further information that you might find helpful if available.
Issues include :
- Occasional inability to detect TSL2561 device. (This could be my own connection.)
- Occasional issue with rounding Lux value resulting in non-integer value used for screen brightness:
```
# Relevant lines from journalctl -u lightsensor.service -b
Nov 23 18:02:35 CRANKSHAFT-NG systemd[1]: Started Crankshaft Lightsensor Service.
Nov 23 18:02:36 CRANKSHAFT-NG service_lightsensor.py[4466]: TSL2561
Nov 23 18:02:36 CRANKSHAFT-NG service_lightsensor.py[4466]: Lux = 4.3 | Level 1 -> trigger night
Nov 23 18:02:36 CRANKSHAFT-NG service_lightsensor.py[4466]: /usr/local/bin/crankshaft: line 691: [: b30: integer expression expected
Nov 23 18:02:36 CRANKSHAFT-NG service_lightsensor.py[4466]: /usr/local/bin/crankshaft: line 697: [: b30: integer expression expected
Nov 23 18:02:36 CRANKSHAFT-NG root[4507]: CSMT-LOGGER: --------------------------------------------------------------------
Nov 23 18:02:36 CRANKSHAFT-NG root[4508]: CSMT-LOGGER: Brightness Level Set To: b30
Nov 23 18:02:36 CRANKSHAFT-NG root[4509]: CSMT-LOGGER: --------------------------------------------------------------------
```
- Occasional issue with communicating with TSL2561 at all.
```
# Relevant lines from journalctl -u lightsensor.service -b
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]: TSL2561
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]: Traceback (most recent call last):
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]:   File "/opt/crankshaft/service_lightsensor.py", line 101, in <module>
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]:     Luxrounded = get_LUX(LIGHTSENSOR)
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]:   File "/opt/crankshaft/service_lightsensor.py", line 34, in get_LUX
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]:     Lux = get_LUX_TSL2561(TSL_I2C_BUS, TSL_ADDR)
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]:   File "/opt/crankshaft/service_lightsensor.py", line 46, in get_LUX_TSL2561
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]:     i2cBus.write_byte_data(TSL_ADDR, 0x80, 0x03)
Nov 23 17:35:24 CRANKSHAFT-NG service_lightsensor.py[1542]: OSError: [Errno 121] Remote I/O error
```

### Debug Info
[debug.zip](https://github.com/opencardev/crankshaft/files/7590034/debug.zip)

techdev5521, 2021-11-23:
This bug is both to report the issue as well as to document the work I am doing to address it.

Novocaine85, 2021-11-23:
I would like to add that the automatic brightness is not working at all with the sensor. On previous release (I don't remember which one, but it was a very old one like 2019) was working good.

matt2005, 2021-11-23:
@techdev5521 I suspect the brightness issue is due to [Line 691](https://github.com/opencardev/prebuilts/blob/master/csmt/crankshaft#L691) and [Line 697](https://github.com/opencardev/prebuilts/blob/master/csmt/crankshaft#L697) as they are erroring as per your issue report
It should be reporting as an Integer but appear to be b30 so it may need a type conversion.

I suspect it got broken when i added the TSL2591 support [TSL2591 support](https://github.com/opencardev/crankshaft/commit/eff477922c809f2ebc1761d9be81c8f9878e8756#diff-6b853a728b3168b212ce4933b8612b54cfb7076b96400072b2854f5caa21190c)

techdev5521, 2021-11-24:
I agree @matt2005 -- my plan is to debug the service script by stopping the service and running the script as a standalone file to see where exactly issues are. I am confident I can resolve the issues with the TSL2561 but I do not have a TSL2591 and may need your help ensuring I don't broke that if you have one. I'll see what i can do later today.

jefftherobot, 2021-12-30:
@techdev5521 My TSL2591 isn't adjusting brightness. I have the same log errors as @techdev5521

jefftherobot, 2021-12-30:
> 

I have TSL2591 and can help debug.

techdev5521, 2022-01-05:
I found the issue in the Python script that handles light sensor values but I'm currently traveling without my hardware or stable Internet. I should be back to both next week and can do some testing / post updates.

matt2005, 2022-02-08:
@techdev5521 have you been able to fix the script?

matt2005, 2022-04-09:
@techdev5521 any update on the pr for the fix?

techdev5521, 2022-04-11:
Yes! I’m back from Colombia finally and can get that in this week. 

Cheers,
Justin

matt2005, 2022-06-25:
Any joy with the pr?

techdev5521, 2022-08-30:
Hey -- sorry for dropping off the planet. I'll make time to work on this this week and submit work. The fix is fairly simple.

CoordinatedEffort, 2022-09-10:
@techdev5521 just wanted to see if you've had a chance to implement a fix.

MikeDotNetty, 2022-12-20:
@techdev5521 just wanted to see if there was any movement on this?   Don't mean to be a hassle and wish you the best, but I would love this simple fix to be implemented if you have the time.

MikeDotNetty, 2022-12-21:
@techdev5521 Thanks, best of luck and happy holidays!

MikeDotNetty, 2022-12-28:
@techdev5521 just wondering if you had a chance over the last few days to look into this?

techdev5521, 2023-01-29:
Hey all. Sitting down now with my dev kit to submit a fix for this.

myfrom, 2023-02-10:
Hi, I have been struggling with this issue as well but I managed to make a workaround and I'm close to finding the source of the problem, I think.

In short, the `crankshaft brightness set xx` command is being called with additional letter "b" in front of the value , which is what causes the error:
`/usr/local/bin/crankshaft: line 697: [: b30: integer expression expected`
My workaround replaces $3 callback with a variable that omits first character from $3, however this isn't optimal as it breaks many things including setting brightness from command line.

I am using sensor TSL25911 (which afaik is almost identical to TSL2591 but produced by Waveshare and has hardware more suitable for my application).

My findings lead my to believe the error most likely originates in [service_lightsensor.py:10](https://github.com/opencardev/crankshaft/blob/66525ef9ce3580dbfdb946c6bb1985aa59b1e07f/stage3/03-crankshaft-base/files/opt/crankshaft/service_lightsensor.py#L10).

`get_var` function returns all values incorrectly:
````py
>>> import subprocess
>>> CMD = 'echo $(source ./crankshaft_env.sh; echo $%s)' % 'LUX_LEVEL_1'
>>> p = subprocess.Popen(CMD, stdout=subprocess.PIPE, shell=True, executable='/bin/bash')
>>> print(p.stdout.readlines()[0].strip())
b'5'
````
Running `int()` on the output corrects it:
````py
>>> import subprocess
>>> CMD = 'echo $(source ./crankshaft_env.sh; echo $%s)' % 'DISP_BRIGHTNESS_1'
>>> p = subprocess.Popen(CMD, stdout=subprocess.PIPE, shell=True, executable='/bin/bash')
>>> print(int(p.stdout.readlines()[0].strip()))
30
>>>
````
This is why there are no errors for example on line 107

**Changing lines 109, 113, 117, 121, 125 and 129 to `int` instead of `str` should fix the issue IMO**. If I'll have time, I'll test my theory by the end of this week.

MikeDotNetty, 2023-02-17:
@myfrom just wondering if you've had any luck with this?

myfrom, 2023-02-17:
I got stuck because my car went to a mechanic and with it the lightsensor setup 🫤 I'm expecting to get it back today or Monday and will work on that as soon as that happens. If you have a running setup and would be willing to test it before that time I can send more info

MikeDotNetty, 2023-02-17:
@myfrom sorry to hear that...trips to the mechanic are never fun.  I do have crankshaft setup in my car on my raspberry pi 3b+, and I have the TSL25911 sensor... I'd be more than happy to test for you.  Just let me know.  Thanks!

MikeDotNetty, 2023-02-17:
Thanks! I'll do that this weekend and report my results.  Much appreciated!

myfrom, 2023-02-17:
Update: I got my car back (yay) and tested the changes right away. In my testing, they do work and lightsensor.service changes screen brightness as expected. PR opened.

(Btw, @MikeDotNetty if you still want to carry on your own tests, I highly recommend editing the file with nano, trying to move the file is more elaborate than I thought, running into permission and chmod issues)

MikeDotNetty, 2023-02-18:
That's great news!  I'll certainly try it this weekend.  And yes I go the nano way and report back.  Thanks again!

MikeDotNetty, 2023-02-18:
@myfrom While I was not able to get your modified code to work.  I think it might be due to the fact that my TSL25911 is really a "TSL25911FN".  I have attached the error.  This error is NOT related to your updated code, but rather it's failing on obtaining the LUX value to work with.  REGARDLESS, you set me on the right path, and I was able to update this service_lightsensor.py file specific to my sensor, such that it **IS WORKING** now.  So thanks for that!

Here's the error in case you are interested:

> Feb 14 05:12:52 CRANKSHAFT-NG systemd[1]: Starting Crankshaft Lightsensor Service...
Feb 14 05:12:52 CRANKSHAFT-NG systemd[1]: Started Crankshaft Lightsensor Service.
Feb 14 05:12:53 CRANKSHAFT-NG service_lightsensor.py[1447]: TSL2561
Feb 14 05:12:53 CRANKSHAFT-NG service_lightsensor.py[1447]: Traceback (most recent call last):
Feb 14 05:12:53 CRANKSHAFT-NG service_lightsensor.py[1447]:   File "/opt/crankshaft/service_lightsensor.py", line 101, in <module>
Feb 14 05:12:53 CRANKSHAFT-NG service_lightsensor.py[1447]:     Luxrounded = get_LUX(LIGHTSENSOR)
Feb 14 05:12:53 CRANKSHAFT-NG service_lightsensor.py[1447]:   File "/opt/crankshaft/service_lightsensor.py", line 39, in get_LUX
Feb 14 05:12:53 CRANKSHAFT-NG service_lightsensor.py[1447]:     return round(Lux, 1)
Feb 14 05:12:53 CRANKSHAFT-NG service_lightsensor.py[1447]: TypeError: type NoneType doesn't define __round__ method
Feb 14 05:12:53 CRANKSHAFT-NG systemd[1]: lightsensor.service: Main process exited, code=exited, status=1/FAILURE
Feb 14 05:12:53 CRANKSHAFT-NG systemd[1]: lightsensor.service: Failed with result 'exit-code'.
Feb 14 05:12:58 CRANKSHAFT-NG systemd[1]: lightsensor.service: Service RestartSec=5s expired, scheduling restart.
Feb 14 05:12:58 CRANKSHAFT-NG systemd[1]: lightsensor.service: Scheduled restart job, restart counter is at 1.
Feb 14 05:12:58 CRANKSHAFT-NG systemd[1]: Stopped Crankshaft Lightsensor Service.
Feb 14 05:12:58 CRANKSHAFT-NG systemd[1]: Starting Crankshaft Lightsensor Service...
Feb 14 05:12:58 CRANKSHAFT-NG systemd[1]: Started Crankshaft Lightsensor Service.
Feb 14 05:12:58 CRANKSHAFT-NG service_lightsensor.py[1523]: TSL2561
Feb 14 05:12:58 CRANKSHAFT-NG service_lightsensor.py[1523]: Traceback (most recent call last):
Feb 14 05:12:58 CRANKSHAFT-NG service_lightsensor.py[1523]:   File "/opt/crankshaft/service_lightsensor.py", line 101, in <module>
Feb 14 05:12:58 CRANKSHAFT-NG service_lightsensor.py[1523]:     Luxrounded = get_LUX(LIGHTSENSOR)
Feb 14 05:12:58 CRANKSHAFT-NG service_lightsensor.py[1523]:   File "/opt/crankshaft/service_lightsensor.py", line 39, in get_LUX
Feb 14 05:12:58 CRANKSHAFT-NG service_lightsensor.py[1523]:     return round(Lux, 1)
Feb 14 05:12:58 CRANKSHAFT-NG service_lightsensor.py[1523]: TypeError: type NoneType doesn't define __round__ method
Feb 14 05:12:58 CRANKSHAFT-NG systemd[1]: lightsensor.service: Main process exited, code=exited, status=1/FAILURE
Feb 14 05:12:58 CRANKSHAFT-NG systemd[1]: lightsensor.service: Failed with result 'exit-code'.
Feb 18 12:46:37 CRANKSHAFT-NG systemd[1]: lightsensor.service: Service RestartSec=5s expired, scheduling restart.
Feb 18 12:46:37 CRANKSHAFT-NG systemd[1]: lightsensor.service: Scheduled restart job, restart counter is at 2.
Feb 18 12:46:37 CRANKSHAFT-NG systemd[1]: Stopped Crankshaft Lightsensor Service.
Feb 18 12:46:37 CRANKSHAFT-NG systemd[1]: Starting Crankshaft Lightsensor Service...
Feb 18 12:46:37 CRANKSHAFT-NG systemd[1]: Started Crankshaft Lightsensor Service.
Feb 18 12:46:37 CRANKSHAFT-NG service_lightsensor.py[1659]: TSL2561
Feb 18 12:46:37 CRANKSHAFT-NG service_lightsensor.py[1659]: Traceback (most recent call last):
Feb 18 12:46:37 CRANKSHAFT-NG service_lightsensor.py[1659]:   File "/opt/crankshaft/service_lightsensor.py", line 101, in <module>
Feb 18 12:46:37 CRANKSHAFT-NG service_lightsensor.py[1659]:     Luxrounded = get_LUX(LIGHTSENSOR)
Feb 18 12:46:37 CRANKSHAFT-NG service_lightsensor.py[1659]:   File "/opt/crankshaft/service_lightsensor.py", line 39, in get_LUX
Feb 18 12:46:37 CRANKSHAFT-NG service_lightsensor.py[1659]:     return round(Lux, 1)
Feb 18 12:46:37 CRANKSHAFT-NG service_lightsensor.py[1659]: TypeError: type NoneType doesn't define __round__ method
Feb 18 12:46:37 CRANKSHAFT-NG systemd[1]: lightsensor.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 12:46:37 CRANKSHAFT-NG systemd[1]: lightsensor.service: Failed with result 'exit-code'.
Feb 18 12:46:42 CRANKSHAFT-NG systemd[1]: lightsensor.service: Service RestartSec=5s expired, scheduling restart.
Feb 18 12:46:42 CRANKSHAFT-NG systemd[1]: lightsensor.service: Scheduled restart job, restart counter is at 3.
Feb 18 12:46:42 CRANKSHAFT-NG systemd[1]: Stopped Crankshaft Lightsensor Service.
Feb 18 12:46:42 CRANKSHAFT-NG systemd[1]: lightsensor.service: Start request repeated too quickly.
Feb 18 12:46:42 CRANKSHAFT-NG systemd[1]: lightsensor.service: Failed with result 'exit-code'.
Feb 18 12:46:42 CRANKSHAFT-NG systemd[1]: Failed to start Crankshaft Lightsensor Service

MikeDotNetty, 2023-02-18:
@myfrom actually, now that I think about it ..without your fix, my fix most likely would not have worked, so thanks for that!

myfrom, 2023-02-18:
Glad I could help and thanks for trying it out 😉

CoordinatedEffort, 2023-08-05:
@MikeDotNetty @myfrom Would either of you mind giving instructions on how to get this working? I understand the changes that need to be made but I'm a Linux newb and I don't know how to actually do it.

CoordinatedEffort, 2023-08-05:
@MikeDotNetty @myfrom I figured out how to SSH and use nano to change the file. However, I noticed that the home screen doesn't display the lux value anymore in the upper left corner and the brightness still doesn't seem to be adjusting automatically.

myfrom, 2023-09-24:
> @MikeDotNetty @myfrom I figured out how to SSH and use nano to change the file. However, I noticed that the home screen doesn't display the lux value anymore in the upper left corner and the brightness still doesn't seem to be adjusting automatically.
> 
> Edit: after rebooting a few times, now it shows initial lux value on startup but then it never updates after that. The GUI's brightness does actually change depending on this initial lux value. I have a TSL2591.
> 
> Edit 2: after a few more reboots, it stopped showing any lux value again and isn't adjusting the GUI's brightness.

I'd check the wiring and sensor first. It sounds like that might be the issue with what you're describing

techdev5521, 2023-09-25:
@CoordinatedEffort I have these issues mapped out with details, files, causes and whatnot but I've not the time to transfer my paper notes to a proper comment. Would you be interested in a voice/video chat sometime to go over that info?

CoordinatedEffort, 2023-09-26:
@techdev5521 I'm definitely not opposed to having a chat and I'm very grateful for the offer of your time, but my technical expertise is basically non-existent. I'd really only be able to follow detailed instructions (as you might be able to tell from my earlier comments) so I'm not sure that I'd have much to offer in a discussion. Is it that you have a possible solution that you'd like me to try?

CoordinatedEffort, 2025-05-12:
Now that there's been recent activity on this project with new builds, I'm wondering if it might be possible to implement the light sensor fixes discussed in this issue?

techdev5521, 2025-05-12:
Certainly! I can dig up my old notes and post something within the week.

techdev5521, 2025-06-10:
Update: I was able to update my dev rig from 2021 to the latest release. I will look into retracing my notes tomorrow night. Now that the rig is setup and updated it should be just a couple of days.

CoordinatedEffort, 2025-06-18:
Sorry for the rather novice question (I'm a novice) but will you be able to test for both the TSL2561 and TSL2591? I have a 2591 and I wasn't sure if that required testing independent of the 2561.

techdev5521, 2025-08-11:
My apologies for repeated delays. I've set aside some time to setup and test this patch after reviewing my notes. I should have a PR ready on Wedneday!

CoordinatedEffort, 2026-01-11:
I unfortunately have zero technical expertise so I wouldn't be able to contribute anything as far as actually solving the problem goes. I'll gladly test potential fixes though.

techdev5521, 2026-01-11:
@CoordinatedEffort Which sensor do you have?

Right after I posted last, I started getting some data but its odd and makes me wonder if my sensor is bonk. Would be helpful to look at the output on your system. I can help you set it up to test.

techdev5521, 2026-01-11:
I am working on building the project and including my changes. Once I can do that I will open a PR. Until then, I am happy to test out these changes manually over a call to make sure the other light sensor works as well.

closed 2026-10-11 by github-actions[bot]