opencardev/crankshaft #521: unfortunately crankshaft stopped when connected to 5inch display

roshanrp5, 2021-12-27:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- Realme x

- Android 10

- latest

- connect 5inch lcd display
- install lcd5show driver using dev mode
- bootup the pi and it crashes
- bootup the pi and it crashes

  Warning: if you have configured wifi passwords they will be blanked by script during creation.
  We can't make absolutly sure you passwords are removed so do a personal check of the files inside the zip.

matt2005, 2021-12-29:
What pi version?

maskaz, 2022-01-13:
I assume it is 5'' waveshare hdmi display?
I had the same issue. 
If you do not need rotation, just add this entries to /boot/config.txt

max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0  
hdmi_drive=1

matt2005, 2022-09-13:
please try [2022.09.12.1](https://github.com/opencardev/crankshaft/releases/tag/v.2022.09.12.1) that should have the correct naming