opencardev/crankshaft #522: Building Docker Image

sjdean, 2022-01-06:
Small problem with the Crankshaft Docker Build:

Setting up python3-colorzero (1.1) ...
Setting up python3-rpi.gpio (0.7.0-0.1~bpo10+4) ...
Setting up python3-spidev (20200602~200721-1~buster) ...
Setting up zip (3.0-11) ...
Setting up python3-gpiozero (1.6.2-1) ...
Setting up python-smbus:armhf (4.1-1) ...
Processing triggers for man-db (2.8.5-2) ...
[18:55:16] End /pi-gen/stage3/03-crankshaft-base/00-packages
[18:55:16] Begin /pi-gen/stage3/03-crankshaft-base/00-run.sh
cp: cannot stat '/pi-gen/prebuilts/udev/51-android.rules': No such file or directory

Any thoughts appreciated.

matt2005, 2022-01-10:
How are you building it? The docker file is used to build the pi image.

mostafaayesh, 2022-04-04:
> Small problem with the Crankshaft Docker Build:
>
> Setting up python3-colorzero (1.1) ... Setting up python3-rpi.gpio (0.7.0-0.1~bpo10+4) ... Setting up python3-spidev (20200602~200721-1~buster) ... [...]

You need to update submodules to get the prebuilts folder. After cloning you can run ```git submodule update --init``` in the cloned directory to get the prebuilts folder and it will have the .rules file.