opencardev/crankshaft #524: Cam_overlay doesn't work

maskaz, 2022-01-07:
Platform:
Raspberry Pi 3

Version checked:

2019-08-08-crankshaft-ng-E2B068B

With OpenGL enabled: * failed to add service - already in use?
With OpenGL Legacy enabled: Cannot open 'shader-MJPG.frag': 2, No such file or directory

Version checked:

2021-09-22-crankshaft-ng-ea9e288
With OpenGL enabled: * failed to add service - already in use?
With "dtoverlay=vc4-fkms-v3d" removed: Cannot open 'shader-MJPG.frag': 2, No such file or directory

maskaz, 2022-01-13:
I have found workaround.
Those entries neeeds to be commented out:

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
dtoverlay=vc4-fkms-v3d

Regards

closed 2022-05-14 by github-actions[bot]

Ramortada, 2023-02-05:
@maskaz from what type of camera were you using please ?