opencardev/crankshaft #525: OpenAuto crashed while startting OpenAuto in EGL Mode RPi 4B

apichlinski, 2022-01-10:
Problem:
Unfortunately, OpenAuto crashed while startting OpenAuto in EGL Mode.
**Hardware:**
Raspberry Pi 4B 2GB RAM
Waveshare LCD Touchscreen 5" 800x480
SD Card 32GB
Xiaomi Mi 9T

**Software:**
Crankshaft OpenCarOS NG Alpha-7.4 2021-09-22

**Settings:**
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0
hdmi_drive=1
gpu_mem=128 and 256 (both tested)
dtoverlay=vc4-fkms-v3d (uncommented and commented)

I can boot in devmode = 1 but then GUI won't start. 
How can i start GUI in debug mode?

matt2005, 2022-01-10:
When in dev mode please run the following and share the output

This will create a debug log archive
```bash
crankshaft debuglog
```

This will attempt to run autoapp with more output
```bash
export QT_DEBUG_PLUGINS=1
GST_DEBUG=3 /usr/bin/autoapp -platform eglfs
```

apichlinski, 2022-01-10:
I have to add that in my case GUI works only for X11 mode.

@matt2005  You can find my debuglog here:
[debug.zip](https://github.com/opencardev/crankshaft/files/7842623/debug.zip)

autoapp response:
`$ autoapp -platform eglf
QFactoryLoader::QFactoryLoader() checking directory path "/usr/local/qt5/plugins/platforms" ...
QFactoryLoader::QFactoryLoader() looking at "/usr/local/qt5/plugins/platforms/libqeglfs.so"
Found metadata in lib /usr/local/qt5/plugins/platforms/libqeglfs.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "eglfs"
        ]
    },
    "archreq": 0,
    "className": "QEglFSIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("eglfs")
QFactoryLoader::QFactoryLoader() looking at "/usr/local/qt5/plugins/platforms/libqlinuxfb.so"
Found metadata in lib /usr/local/qt5/plugins/platforms/libqlinuxfb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "linuxfb"
        ]
    },
    "archreq": 0,
    "className": "QLinuxFbIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("linuxfb")
QFactoryLoader::QFactoryLoader() looking at "/usr/local/qt5/plugins/platforms/libqminimal.so"
Found metadata in lib /usr/local/qt5/plugins/platforms/libqminimal.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimal"
        ]
    },
    "archreq": 0,
    "className": "QMinimalIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("minimal")
QFactoryLoader::QFactoryLoader() looking at "/usr/local/qt5/plugins/platforms/libqminimalegl.so"
Found metadata in lib /usr/local/qt5/plugins/platforms/libqminimalegl.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimalegl"
        ]
    },
    "archreq": 0,
    "className": "QMinimalEglIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("minimalegl")
QFactoryLoader::QFactoryLoader() looking at "/usr/local/qt5/plugins/platforms/libqoffscreen.so"
Found metadata in lib /usr/local/qt5/plugins/platforms/libqoffscreen.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "offscreen"
        ]
    },
    "archreq": 0,
    "className": "QOffscreenIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("offscreen")
QFactoryLoader::QFactoryLoader() looking at "/usr/local/qt5/plugins/platforms/libqvnc.so"
Found metadata in lib /usr/local/qt5/plugins/platforms/libqvnc.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "vnc"
        ]
    },
    "archreq": 0,
    "className": "QVncIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("vnc")
QFactoryLoader::QFactoryLoader() looking at "/usr/local/qt5/plugins/platforms/libqxcb.so"
Found metadata in lib /usr/local/qt5/plugins/platforms/libqxcb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "xcb"
        ]
    },
    "archreq": 0,
    "className": "QXcbIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/usr/local/bin/platforms" ...
qt.qpa.plugin: Could not find the Qt platform plugin "eglf" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted
`

matt2005, 2022-01-12:
You missed the s off eglfs

mayurkaul, 2022-07-29:
i have the similar issue with the exact same scenario. i did all the things that @apichlinski did too. But no help. i'm still getting  the dreaded

`Unfortunately, OpenAuto crashed while startting OpenAuto in EGL Mode.`

have attached my [debug.zip](https://github.com/opencardev/crankshaft/files/9216434/debug.zip) too

I have an EIZO display connected via HDMI. and i'm on Crankshaft OpenCarOS NG alpha-7.5 2022-06-29. Please help

matt2005, 2022-09-12:
please try [2022.09.12.1](https://github.com/opencardev/crankshaft/releases/tag/v.2022.09.12.1) that should have the correct naming

AntiGator, 2023-04-20:
> please try [2022.09.12.1](https://github.com/opencardev/crankshaft/releases/tag/v.2022.09.12.1) that should have the correct naming
i have the similar issue to.
No change with this release again OpenAuto crashed while startting OpenAuto in EGL Mode.

closed 2025-01-12 by matt2005