opencardev/crankshaft #536: Website: Discontinue the use of Google Photos

htruong, 2022-02-07:
I no longer am active in the Crankshaft community due to personal reasons.

The Crankshaft pictures links to https://photos.app.goo.gl/81hQ6wTuLFNGmRHh2. This is my personal account's album that allows people to join. I created a long time ago to facilitate the open sharing of Crankshaft installations.

Recently, Google no longer allows us to store standard-res photos on their platform, making this a very unattractive place to share pictures, as people need to pay to get into this (closed) platform. I am going to discontinue the album at some point in the future. It will be not be before July of this year, but I have no promise that it will be accessible after that. So if someone could suggest a better solution and help moving this to a new platform, that'd be much appreciated.

techdev5521, 2022-02-08:
Could the images be placed directly in the GitHub repo?

htruong, 2022-02-08:
We can but that'd be hard to add more pictures and add comments.

techdev5521, 2022-02-08:
Ah -- I see what you mean. You want to keep both the pictures and the ability to comment. I would be willing to host a public image board. There's a self hosted software gist somewhere i can go to look for a software solution to this and set it up. It could be accessible under the Crankshaft domain too.

matt2005, 2022-02-08:
maybe a jekyll hosted blog using github pages?

matt2005, 2022-06-14:
Still investigating jeykll solution

matt2005, 2022-06-29:
I've copied the album as a stepping stone while a better solution is figured out

closed 2022-09-11 by github-actions[bot]