opencardev/crankshaft #567: Raspberry Pi 2B

atomdude, 2022-05-23:
i followed installation instructions, tried to install on Raspberry 2B and it failed saying that failed to start auto......

i read that crankshaft should work on raspeberry version 2B any ideas why?

matt2005, 2022-06-03:
Could you try this image [pi2 image](https://github.com/opencardev/crankshaft/releases/download/untagged-abf8030d8708ae211544/2022-02-04-crankshaft-ng-496df4a-pi2.zip)

lazyjavvy, 2022-09-06:
@matt2005 Thank you very much for working on this issue.
The latest image for pi 2 (2022-02-04-crankshaft-ng-496df4a-pi2.zip) works for me after following the instructions for updating QT at https://github.com/opencardev/crankshaft/issues/450#issuecomment-1008029456
It seems that the Qt binaries in the release image are still for ARMv8.

matt2005, 2022-09-07:
Ah, i think the build has mixed up the pi2 pi3/4 QT libraries. Can you try the other image? @lazyjavvy

lazyjavvy, 2022-09-08:
The other image works.

robertw, 2022-09-10:
could you please show me which pi 2 image works thank you

lazyjavvy, 2022-09-10:
[2022-02-04-crankshaft-ng-496df4a.zip](https://github.com/opencardev/crankshaft/releases/download/csng-alpha7.5/2022-02-04-crankshaft-ng-496df4a.zip) works for pi 2 model B v1.1

matt2005, 2022-09-10:
I'm building the new images. They should be ready tomorrow. Hopefully the fix has resolved the filename mixup

robertw, 2022-09-11:
thank you guys got it working pi 2 !

matt2005, 2022-09-12:
please try [2022.09.12.1](https://github.com/opencardev/crankshaft/releases/tag/v.2022.09.12.1) that should have the correct naming

atomdude, 2022-10-11:
ok i will try it, i am travelling now and will do in the next couple of weeks.

Thanks