opencardev/crankshaft #583: OpenAuto crashed with RPi 3B+

agoNITE, 2022-08-13:
- What phone do you have?
N/A

- What crankshaft version?
[Crankshaft OpenCarOS NG alpha-7.5 2022-06-29](https://github.com/opencardev/crankshaft/releases/tag/csng-alpha7.5)

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Flash the image to a SD card, and boot it. Yes.

After boot, I get the following error:

```
Unfortunately, OpenAuto crashed.
Please report this indicident to Crankshaft
Restart in 5 seconds...
```

I'm using the original Raspberry Pi 7" touchscreen.

- Debug logs (both EGL and X11)
 [debug_EGL.zip](https://github.com/opencardev/crankshaft/files/9331073/debug_EGL.zip)
[debug_X11.zip](https://github.com/opencardev/crankshaft/files/9331074/debug_X11.zip)

mario-correa, 2022-08-22:
same config here, and same error

oakwoodMatt, 2022-08-23:
Same config and error here. Cannot boot.

sylwesterszczepaniak, 2022-09-02:
Same error. Perhaps I can help debug/fix?

mario-correa, 2022-09-03:
I fix it by downloading a previous release. It worked as expected.

nputman, 2022-09-04:
same issue

matt2005, 2022-09-04:
which image are you using? the release has 2 images.
If you get a chance could you try the pi2 image, as the images may have gotten mixed up during the build.

nputman, 2022-09-04:
The pi2 image booted up and seems to work

matt2005, 2022-09-07:
Great, i'm fixing the build issue and then I'll issue an updated image.

agoNITE, 2022-09-09:
> Great, i'm fixing the build issue and then I'll issue an updated image.

Great! Luckily I didn't have any time to flash the previous build. So I'll be patiently waiting for the updated image, thanks!

matt2005, 2022-09-12:
please try [2022.09.12.1](https://github.com/opencardev/crankshaft/releases/tag/v.2022.09.12.1) that should have the correct naming

sylwesterszczepaniak, 2022-09-13:
Works out of the box. Thank you!

closed 2023-01-12 by github-actions[bot]