opencardev/crankshaft #594: Night mode brightness issues with official 7" rpi screen

atroph, 2022-09-21:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have? Pixel 4 XL

- What OS version? 13

- What crankshaft version? 2022-09-11-crankshaft-ng-66525ef.zip
- rpi3
- 7" official touchscreen 

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible.

*this is a copy of #495 that was closed due to inactivity. 

Change day/night mode brightness settings in config menu to something different. 
Select day/night and observe changes in log. 
Screen blanks and then restores brightness to a level that isn't correct. 
If number has too low of last two digits the screen with blank, appear to come up, and then blank once again. 
-ie: 100, 200 will result in screen brightness of 0. 
It appears that the first digit is removed from the variable and only the last two are saved/restored from temp location
Set to 255 day and 220 night so screen doesn't go completely dark. 
-Result 55 day 20 night
Can reproduce with 155 and 120 with the same results when the brightness is restored. 
-Result 55 day 20 night

In the log below the correct numbers as per the config are 255 and 220 for day/night respectively. 

- Please provide any further information that you might find helpful if available.

Sep 20 23:50:56 CRANKSHAFT-NG pi: SERV-LOGGER: Caller: /opt/crankshaft/service_daynight.sh
Sep 20 23:50:56 CRANKSHAFT-NG pi: SERV-LOGGER: Restore brightness night by app
Sep 20 23:50:56 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:56 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:56 CRANKSHAFT-NG pi: CSMT-LOGGER: Start Restore Brightness Level Night - Current: 255
Sep 20 23:50:56 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:57 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:57 CRANKSHAFT-NG pi: CSMT-LOGGER: Brightness Level Night Restored: 220
Sep 20 23:50:57 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:58 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:58 CRANKSHAFT-NG pi: SERV-LOGGER: Caller: /opt/crankshaft/service_daynight.sh
Sep 20 23:50:58 CRANKSHAFT-NG pi: SERV-LOGGER: Restore brightness day by app
Sep 20 23:50:58 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:59 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:59 CRANKSHAFT-NG pi: CSMT-LOGGER: Start Restore Brightness Level Day - Current: 20
Sep 20 23:50:59 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:59 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:50:59 CRANKSHAFT-NG pi: SERV-LOGGER: Caller: /opt/crankshaft/service_bluetooth.sh
Sep 20 23:50:59 CRANKSHAFT-NG pi: SERV-LOGGER: Bluetooth auto connect attempt: 4
Sep 20 23:50:59 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:51:02 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:51:02 CRANKSHAFT-NG pi: CSMT-LOGGER: Brightness Level Day Restored: 255
Sep 20 23:51:02 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:13 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:13 CRANKSHAFT-NG pi: SERV-LOGGER: Caller: /opt/crankshaft/service_daynight.sh
Sep 20 23:52:13 CRANKSHAFT-NG pi: SERV-LOGGER: Restore brightness night by app
Sep 20 23:52:13 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:13 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:13 CRANKSHAFT-NG pi: CSMT-LOGGER: Start Restore Brightness Level Night - Current: 55
Sep 20 23:52:13 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:15 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:15 CRANKSHAFT-NG pi: CSMT-LOGGER: Brightness Level Night Restored: 220
Sep 20 23:52:15 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:16 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:16 CRANKSHAFT-NG pi: SERV-LOGGER: Caller: /opt/crankshaft/service_daynight.sh
Sep 20 23:52:16 CRANKSHAFT-NG pi: SERV-LOGGER: Restore brightness day by app
Sep 20 23:52:16 CRANKSHAFT-NG pi: SERV-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:17 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:17 CRANKSHAFT-NG pi: CSMT-LOGGER: Start Restore Brightness Level Day - Current: 20
Sep 20 23:52:17 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:19 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------
Sep 20 23:52:19 CRANKSHAFT-NG pi: CSMT-LOGGER: Brightness Level Day Restored: 255
Sep 20 23:52:19 CRANKSHAFT-NG pi: CSMT-LOGGER: --------------------------------------------------------------------

atroph, 2022-09-21:
It also appears that when watching the brightness switch there is a "fade up" or "fade down" when changing the brightness. 

This process flickers which makes me believe that the variable for this function is overflowing between 0 and 99 as the screen goes blank then settles on the number desired minus the first digit. 

This can also be observed by:
-Tap the day/night button
-Immediately afterwards hit the brightness button multiple times and watch the value when the brightness is changing. 
-It varies between 0-99 before it settles on the final number

Bad hex to dec conversion or mathematical operation?

lazyjavvy, 2022-09-26:
I observed the same issue on raspberry pi 2B with the official 7 inch touch screen. My temporarily workaround is to disable the "fade up" and "fade down" effects by replacing the while loops in /usr/local/bin/crankshaft from line 491 to 503 with `echo ${BR_NIGHT} > ${BRIGHTNESS_FILE}` and line 519 to 531 with `echo ${BR_DAY} > ${BRIGHTNESS_FILE}`.

closed 2023-01-25 by github-actions[bot]