opencardev/crankshaft #635: Android Auto Wireless Dongle

mancioshell, 2023-07-15:
Is It possibile to use a Raspberry with this software to create an USB Dongle attached to a native car headunit in order to use Android Auto in wireless mode with a smartphone?

rorosaurus, 2023-09-18:
+1 I'm also interested in using Android auto wirelessly. Is a USB cable required for Crankshaft or can it connect via wifi/Bluetooth like other 3rd party head units?

parkerlreed, 2023-09-18:
Two things going on here...

> Is It possibile to use a Raspberry with this software to create an USB Dongle attached to a native car headunit in order to use Android Auto in wireless mode with a smartphone?

No. Crankshaft and by extension the upstream AASDK does not support being "pass through" and pretending to be an Android Auto device itself.

> +1 I'm also interested in using Android auto wirelessly. Is a USB cable required for Crankshaft or can it connect via wifi/Bluetooth like other 3rd party head units?

Yes. You would have to do the Android Auto developer options and enable the headunit server. Then you can give the IP of the phone to the main OpenAuto application and it will connect. Nothing as seamless as a headunit designed for this though.

There's the third option of Headunit Reloaded which is created by the AAWireless guys. You can use other Android devices as a P2P connection that gets much closer to the full integrated experience. They also sell a wireless dongle that would solve the first issue as well.

closed 2024-01-17 by github-actions[bot]