opencardev/crankshaft #668: Problem with 4.3inch DSI LCD  touchscreen

chingchong879, 2024-05-24:
- What phone do you have? Zenfone9

- What OS version?  Android 14

- What crankshaft version? 2022-09-11 v

- What pi version? raspberry 4 pi

my screen is 4.3inch DSI LCD 800×480
When I inserted the SD card into raspberry and it installed successfully, but I couldn't get into the system, and the screen stayed stuck on black.
and i add this in to config.txt
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0
hdmi_drive=1
but still won't work

Geekyadz, 2024-06-16:
What one do you have specifically? My one ( https://thepihut.com/products/4-3-dsi-capacitive-touchscreen-display-for-raspberry-pi-800x480?variant=41483994300611 ) has been working with changing any files

chingchong879, 2024-06-16:
yes it's this one

chingchong879, 2024-06-16:
maybe my sd is broken

Geekyadz, 2024-06-16:
I'm not convinced it's your SD card. 

What I'm more convinced about is that it's the actual DSI cake itself or the lines you have added. 

Simple test: flash crankshaft to the SD card again and see if that works without adding anything to the files

chingchong879, 2024-06-16:
> What I'm more convinced about is that it's the actual DSI cable itself or

okay i will try it later

Geekyadz, 2024-06-23:
Any update on this?

chingchong879, 2024-08-10:
hey can i add you on discord? Geekyadz

chingchong879, 2024-08-10:
> I'm not convinced it's your SD card.
> 
> What I'm more convinced about is that it's the actual DSI cable itself or the lines you have added.
> 
> Simple test: flash crankshaft to the SD card again and see if that works without adding anything to the files

no cause i can see some system install but after that just stuck in black screen

Geekyadz, 2024-08-18:
> no cause i can see some system install but after that just stuck in black screen

Then yes it does sound like it's the SD card if it's not even installing but you can see it doing its thing (ie the text)

closed 2024-12-17 by github-actions[bot]