opencardev/crankshaft #688: Issue with cam_overlay in EGL mode

georgi-m-iliev, 2024-11-02:
Hello,

I am not sure if this is the right place to ask my question, but at least I'd gather attention. I am trying to run Crankshaft in EGL mode and have a reverse camera at the same time. Of course this isn't going as planned. I am using an AV to USB dongle, which supports YUYV. The issue comes when starting cam_overlay: 
```
Opened device '/dev/video0'
Dimensions: 480 x 320 Pixel Format: YUYV
* failed to add service - already in use
```
I have tried to understand the inner workings of Crankshaft and cam_overlay and came to the conclusion that EGL mode needs to use the GPU of the RPi, so I need `dtoverlay=vc4-fkms-v3d ` for it to work, but from what I've seen on the topic it also breaks cam_overlay for some reason. I even tried building it myself - the main and cpp branch, and failed...

So I guess my question is - Is it possible to run Crankshaft in EGL mode and have cam_overlay work at the same time?

Thanks in advance! I am attaching a debug.zip if anyone is interested.
[debug.zip](https://github.com/user-attachments/files/17608213/debug.zip)

Geekyadz, 2025-01-12:
What are you using to run crankshaft?

georgi-m-iliev, 2025-01-12:
Pi Zero 2 W and a Waveshare Zero-DISP-7A.

Angus-C-git, 2025-01-19:
I run into the same issue here :(

closed 2025-06-19 by github-actions[bot]