opencardev/crankshaft #700: Problem with image on Rsapberry 3 B+

trybikowski, 2025-01-06:
Hi,

I’m encountering a problem while trying to install the image on my Raspberry Pi 3 B+. Below are the details of the issue:

1. During the installation, I encountered this error:
[image]

2. After booting, the system displays the following screen:
[image]

I am using the following setup:

Device: Raspberry Pi 3 B+
Image: 20250105-Debian_buster-armhf.zip (installed on an SD card)

SonOfGib, 2025-01-06:
I am seeing the same thing, I'll look into it.

matt2005, 2025-01-06:
I've switched to a different process to update the compiled binaries to speed up the build time, but i missed a step in the process.
I've added the ldconfig step and now it is working. I'm building a new release so it should be ready in the next 30mins

SonOfGib, 2025-01-06:
I will check once its built, but I think the issue is that the autoapp binary in prebuilts is expecting `libaasdk.so.3` but that symlink is not present in /usr/local/lib. I checked the version I built a few months ago and that link _is_ present. I must have modified the crankshaft pigen script and forgot.

At least, running ldconfig doesn't seem to solve it on the test image I burned.

matt2005, 2025-01-07:
It's not fixed but it's not the ldconfig issue

SonOfGib, 2025-01-07:
I actually just burned 20250107 to an sd card and its working, despite what I thought above. Are you seeing issues?

matt2005, 2025-01-07:
I tested on a pi2, but that might be a separate issue. I'll dig out the pi3 and try that one.

matt2005, 2025-01-12:
@trybikowski this should be resolved in the latest pre-release

trybikowski, 2025-01-14:
Hi @matt2005,

I’ve already installed the latest image (from January 11). The fault after installation still exists, but the image works. Android Auto starts properly.

In my opinion, the issue with the light sensor (TL2561) still persists. It seems to "read" the lux value at startup, but the state doesn’t change afterward.

I’ve also assembled the RTC DS3231 time chip (insted TL2561). While the time is remembered, Crankshaft doesn’t adjust between day and night based on the settings (day: 8:00, night: 17:00). I have to manually switch between day and night using the button in the main menu. Is this ok?

matt2005, 2025-01-14:
Please raise a separate issue for the light sensor issue. Please remember to include hardware details and the debug logs.

PR #701 (2025-01-06): fix ldconfig
fixes #700