opencardev/crankshaft #713: EGL Crashing on all versions > v.2022.09.12.1

Angus-C-git, 2025-01-13:
If you need support: Please post on https://www.reddit.com/r/crankshaft/ as this is a bug tracker, not a support forum.

Note: Reporting issues not based on official published builds will be closed without any comment.

- What phone do you have? Tested on all
   - Pixel 7a
   - Pixel 4a
   - Samsung S23

- What OS version?
   -  Android 15, Kernel 5.10.214

- What crankshaft version?
   - All versions > [v.2022.09.12.1](https://github.com/opencardev/crankshaft/tree/v.2022.09.12.1)

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
   - Flash effected versions and run on **Raspberry PI 4**
   - EGL will crash on startup

- Please provide any further information that you might find helpful if available.

Testing on version [v.2022.09.12.1](https://github.com/opencardev/crankshaft/tree/v.2022.09.12.1) CS will boot just fine but will fail to start android auto UI after detecting a connected device. This happens both with wireless (hotspot) and client modes. 

On [v.2022.09.12.1](https://github.com/opencardev/crankshaft/tree/v.2022.09.12.1) I followed the steps to update online specifically  running:

```bash
crankshaft update check
crankshaft update csmt
csmt update openauto
```
caused EGL to begin crashing too. This leads me to assume that the latest `openauto` binaries/deps (?) are causing the issue.

matt2005, 2025-01-13:
Ah, I suspect the updater is not getting latest aasdk. I'll have a look this evening.

Lotusvball, 2025-01-13:
Having same issue.  Tested on S23 on a Raspberry PI 3 B+

matt2005, 2025-01-13:
Is that with a fresh pre-release image?

Lotusvball, 2025-01-13:
Yes.  Reimaged a microSD card.  Expanded the size and crashed trying to start openauto.

SonOfGib, 2025-01-13:
@Lotusvball @Angus-C-git Does it work in x11 mode? Which image did you experience the issue with? I can't replicate this with the buster-armhf image on my rpi3b+ or pi4.

`<version>-Debian_buster-arm7-pi2.zip`
or
`<version>-Debian_buster-armhf.zip`

Angus-C-git, 2025-01-13:
Thanks all for follow ups I will try supply logs and the other requested info this evening 🍻

 I didn't try the <version>-Debian_buster-armhf.zip image or x11 mode yet.

matt2005, 2025-01-13:
Please try the latest pre-release image.
[20251111](https://github.com/opencardev/crankshaft/releases/tag/20250111)

Angus-C-git, 2025-01-15:
## Testing on [20251111](https://github.com/opencardev/crankshaft/releases/tag/20250111)

In summary I can get a working configuration with the following setup:

1. Disable developer mode for AA on my Pixel 7a
2. Use `20250111-Debian_buster-armhf.img` with [20251111](https://github.com/opencardev/crankshaft/releases/tag/20250111)
3. Plug phone in with default CS settings

I experienced unreliable behaviour in the other configurations but I have a feeling the power supply I was using to my PI was not enough so I changed it part way through testing. I won't push my luck as I'm just pleased to have it working - thanks for all your help. I will revisit this again if I run into issues/other people are still affected. 

### `20250111-Debian_buster-armhf.img`

- [X] Boots 
- [X] EGL Works
- [ ] ~~x11 Mode~~ (skipped)
- [X] AA Client Mode (plugged in)
- [ ] AA Works Wireless Mode
   - Crashes 

### `20250111-Debian_buster-arm7-pi2.img`

- Skipped

Lotusvball, 2025-01-15:
> Please try the latest pre-release image. [20251111](https://github.com/opencardev/crankshaft/releases/tag/20250111)

Sorry, I should have posted earlier that I did us this pre-release.   I used this image:  20250111-Debian_buster-arm7-pi2.zip.

During startup the message I received was openauto failed to open.  I do not have logs but I can try to get some this weekend.

matt2005, 2025-01-15:
Ah I think you've used the wrong image. The pi2 image is for the pi2 use the other image for rpi3/4

closed 2025-02-09 by Angus-C-git