opencardev/crankshaft #717: Extreme Audio stuttering with wireless connection on RPI4

0Flush0, 2025-01-27:
### Issue 

I have installed the newest version of crankshaft ng on my rpi4.
The device starts up in hotspot mode and when connecting to the wifi, android auto wireless works.
Touch and everything is working, altho more laggy than with using a cable. What makes it unusable, is the great amount of lag with spotify playback. 
The audio stutters and lags behind, sometimes playing faster then breaking down. Audio works fine with cable connection.

### Setup

1. RPI 4
2. HDMI Touchscreen 7 inch 1024x600
3. Aux connection to the speakers

### What have i tried

1. Increasing CPU and GPU clock, also increased video memory.
2. Changed wifi mode to 5ghz
3. Used different phones with battery optimizations disabled
4. Client mode is just not working for me: Wifi0 interface is down, couldnt fix it via ssh and manually connecting: Some file not found error

My suspicion is some kind of wifi bottleneck.
Thanks in advance!

pswiifanwmi, 2025-02-05:
I had this issue too, intermittently.

What I mean by that is if my car and my Pi have been off for several hours, and I start Android Auto up, it behaves relatively normally.  If I turn the Pi off and on again after a couple minutes, I get the same scenario as you've described.  What I noticed by checking my phone settings, is that when I'm experiencing the lagging, the phone is actually connected to the Pi in Wifi 4 mode instead of Wifi 5 as I had programmed.

I have since replaced this setup with a $60 Android Auto/Carplay device from China.  It works well enough and operates in Wifi 4 with none of the lagging.  Audio quality is worse, but it doesn't lag like Crankshaft on the Pi 4 does (and it boots in half the time).

All of this is to say that this device does something that Crankshaft does not that makes running in WiFi 4 more consistent.  I suspect there is a lower quality audio codec running that takes up less bandwidth.

I do wonder how a WiFi 6 adapter would fare with Crankshaft, and if there would be problems similar to those with the integrated adapter.  The kernel version that currently ships with Crankshaft NG doesn't support any WiFi 6 adapters, so unless there is a major update to the OS, there is nothing we can do to test it.

viletuna, 2025-02-05:
@pswiifanwmi 
> I do wonder how a WiFi 6 adapter would fare with Crankshaft, and if there would be problems similar to those with the integrated adapter. The kernel version that currently ships with Crankshaft NG [...]

Matt replied in a discussion post I made saying he's planning on updating to the latest OS

devdetj, 2025-02-21:
Something similar happened to me and I solved it. Configuration, audio, audio output, and there you can select the aux output, by default the HDMI output is on.

closed 2025-07-22 by github-actions[bot]