opencardev/crankshaft #748: Raspberry Pi 5 (rpi5) Support

Swivelgames, 2025-07-18:
Wanted to revive this discussion:

- #652
- #671 

It would be great to see Raspberry Pi 5 support added! I'd be interested in helping to contribute, but would love some assistance in direction.

Are there any plans to add Pi 5 support?

matt2005, 2025-07-19:
Yes I'm working on a build

ryantiger658, 2025-08-09:
I am considering doing a new build and would love to build off of the Pi5. How far out are we looking for Pi5 support?

matt2005, 2025-10-21:
I'm working on a Trixie build on a pi5 on 64bit.
I've got aasdk building and openauto running.

I'm in the process of going through all the scripts and services and updating the build.

It's slow going but I'm making progress.