opencardev/crankshaft #751: Raspberry pi3B and official 7" Raspberry pi Touch display2

sixstackjack, 2025-08-01:
i have just got myself a 7" touch display for my raspberry, and i try to get it to work with crankshaft, so i connect the screen and install the Crankshaft OS, and start it up, and the screen stay black, nothing happends what so ever, 

so just to see if the screen was bad or something i installed RaspberrianOS and tryed and the screen worked as intended

so i once again tryed to install Crankshaft-ng, but still no luck with the screen, CrankShaft only seem to work when connecting it directly to a HDMI screen 

anyone have any tips/idea/fixes for my problem

sixstackjack, 2025-08-01:
damn, since im a noob i just want something that i can run from a flash drive, but this seem to be alot more complicated lol

sixstackjack, 2025-08-01:
yea i   did run the "install script" and im still installing, it, but do i need to do some Autorun script aswell or will it start automaticly when i restart my Raspberry, like i sed im total beginner so have no idea how things work :)

sixstackjack, 2025-08-01:
btw, do you have any idea how long i could run the raspberry pi3 with the official 7" screen running in the car before battery drains?

sixstackjack, 2025-08-01:
hmm.. coas my plan is using a converter that turn 12v into a 5v,3a USB coas when i do a measure on my RP3 when its plugged into the wall it measures 0.050A, so its confusing as hell lol

viletuna, 2025-08-01:
I'm assuming that measurement is when the pi is at idle and not underload. You should measure again while it's playing back YouTube or running opendsh and running AA you'll likely get a more accurate reading.

You could always find a switched 12v power supply in your vehicle. Some have the cigarette lighters as switched 12v so you could just pull from that. If not do a quick Google search of your model

sixstackjack, 2025-08-01:
i dunno whats wrong, but i tryed to install the Dash thing twice now and each time it cancels with a "Fatal error: Killer Signal terminated program, cc1plus".... worst part it pops up after the installation have been running for like 3hours or something :(

sixstackjack, 2025-08-01:
i think i figured it out, since i only have 1gb mem, i did increase the Swap files from 512 to 2048mb, and now i maybee can install it, but we shall see  :P

closed 2025-12-31 by github-actions[bot]